LDMX-Software/ldmx-sw #18: *** Error in `ldmx-sim': free(): invalid pointer: 0x0000000000658da0 ***

omar-moreno, 2016-10-25:
The sim application seg faults after it concludes.  The backtrace is as follows: 
 
```
*** Error in `ldmx-sim': free(): invalid pointer: 0x0000000000658da0 ***
======= Backtrace: =========
/lib64/libc.so.6(+0x7364f)[0x7fa8873fd64f]
/lib64/libc.so.6(+0x78eae)[0x7fa887402eae]
/lib64/libc.so.6(+0x79b87)[0x7fa887403b87]
ldmx-sim[0x42af4e]
ldmx-sim[0x42afc4]
ldmx-sim[0x42cb42]
ldmx-sim[0x415f40]
/lib64/libc.so.6(__libc_start_main+0xf5)[0x7fa8873abbe5]
ldmx-sim[0x415e39]

```
The error can be reproced by running the command: 

`ldmx-sim macro`

The macro commands used are as follows: 

```
/persistency/gdml/read detectors/ldmx-det-full-v0/detector.gdml
/run/initialize
/gun/particle e-
/gun/energy 4 GeV
/gun/position -27.926 5 -700 mm
/gun/direction 0.3138 0 3.9877 GeV
/run/beamOn 10
```
The output ROOT file is written correctly.

JeremyMcCormick, 2016-11-03:
If I remove line 26 from the RunManager which deletes the pluginManager then there is no Seg Fault.

JeremyMcCormick, 2016-11-15:
I fixed this on the master by removing deletion of the plugin manager from the run manager.

I will implement a more permanent fix on another issue which includes more restructuring of the plugin manager code.

closed 2016-11-15 by JeremyMcCormick