LDMX-Software/ldmx-sw #37: Sim output file is huge.

JeremyMcCormick, 2016-11-03:
A 50k event sim output file from 4 GeV electrons is ~13 gig which is about 260k per event.

JeremyMcCormick, 2016-11-05:
**Collection and object sizes from the tree**

root [22] t->Print()                                                                                                                                         
******************************************************************************                                                                               
*Tree    :LDMX_Event: LDMX event tree                                        *                                                                               
*Entries :        1 : Total =           89943 bytes  File  Size =      18302 *                                                                               
*        :          : Tree compression factor =   1.00                       *                                                                               
******************************************************************************                                                                               
*Branch  :Event                                                              *                                                                               
*Entries :        1 : BranchElement (see below)                              *                                                                               
*............................................................................*                                                                               
*Br    0 :fUniqueID : UInt_t                                                 *                                                                               
*Entries :        1 : Total  Size=        764 bytes  File Size  =         86 *                                                                               
*Baskets :        1 : Basket Size=      32000 bytes  Compression=   1.00     *                                                                               
*............................................................................*                                                                               
*Br    1 :fBits     : UInt_t                                                 *                                                                               
*Entries :        1 : Total  Size=        756 bytes  File Size  =         94 *                                                                               
*Baskets :        1 : Basket Size=      32000 bytes  Compression=   1.00     *                                                                               
*............................................................................*                                                                               
*Br    2 :eventNumber_ : Int_t                                               *                                                                               
*Entries :        1 : Total  Size=        785 bytes  File Size  =         89 *                                                                               
*Baskets :        1 : Basket Size=      32000 bytes  Compression=   1.00     *                                                                               
*............................................................................*                                                                               
*Br    3 :run_      : Int_t                                                  *                                                                               
*Entries :        1 : Total  Size=        729 bytes  File Size  =         81 *                                                                               
*Baskets :        1 : Basket Size=      32000 bytes  Compression=   1.00     *                                                                               
*............................................................................*                                                                               
*Br    4 :timestamp_ : Int_t                                                 *                                                                               
*Entries :        1 : Total  Size=        771 bytes  File Size  =         87 *                                                                               
*Baskets :        1 : Basket Size=      32000 bytes  Compression=   1.00     *                                                                               
*............................................................................*                                                                               
*Br    5 :weight_   : Double_t                                               *                                                                               
*Entries :        1 : Total  Size=        758 bytes  File Size  =         88 *                                                                               
*Baskets :        1 : Basket Size=      32000 bytes  Compression=   1.00     *                                                                               
*............................................................................*                                                                               
*Br    6 :simParticles_ : Int_t simParticles__                               *                                                                               
[... cut ...]
*Entries :        1 : Total  Size=        947 bytes  File Size  =        105 *
*Baskets :        1 : Basket Size=      32000 bytes  Compression=   1.00     *
*............................................................................*
*Br   95 :targetSimHits_.simParticle_ : TRef simParticle_[targetSimHits__]   *
*Entries :        1 : Total  Size=        996 bytes  File Size  =        112 *
*Baskets :        1 : Basket Size=      32000 bytes  Compression=   1.00     *
*............................................................................*

JeremyMcCormick, 2016-11-05:
collection and object sizes from event

http://www.slac.stanford.edu/~jeremym/ldmx/root_tree_huge.txt

Single float field is 955 bytes, etc.

JeremyMcCormick, 2016-11-07:
Created a [help thread in the ROOT forum](https://root.cern.ch/phpBB3/viewtopic.php?f=3&t=22643).

JeremyMcCormick, 2016-11-09:
Unfortunately, it looks like the events themselves have a reasonable size based on the number of objects in the collections.

There are about 890 ecal hits per event, so assuming a SimCalorimeterHit is ~60 bytes, this is 54k per event in just ecal hit data (uncompressed).  Total event size seems to be around 40k.

Probably we will need to reduce the number of cal hits by performing digitization in the simulation or some similar strategy.

awhitbeck, 2016-11-09:
I can implement an ecal digi processor to at least try to access how much we can reduce the event size, if you think this will help.

JeremyMcCormick, 2016-11-09:
@awhitbeck I think there is some other problem that is ballooning the files by 6-7 times what I would expect.  I calculated that event size should be about 40 kb with the ecal hits but even then 50k events would be ~2 gig but it comes out to ~14 gig or more.  So something is not right.

omar-moreno, 2016-11-09:
@awhitbeck An Ecal digi processor would still be good to implement.  In the end, I don't think we want hundreds of Ecal hits per event.

JeremyMcCormick, 2016-11-14:
This looks much better in the current master.  The Event object is immediately cleared after the Fill() is called on the tree.  This seemed to fix the memory leak that was causing this (maybe?).  I ran 10k events and the file size is 626M which is still pretty big but about 7x smaller than before so it is much more manageable now.

closed 2016-11-14 by JeremyMcCormick