LDMX-Software/ldmx-sw #49: When biasing, only events which are fully processed should be written out.

omar-moreno, 2016-11-09:
Currently, even if an event doesn't pass the biasing conditions, the MC particles and sim hits are still written to the ROOT file.  A user should be able to control which events are written out.

JeremyMcCormick, 2016-11-09:
Probably the cleanest way to do this is raising a G4Exception in your biasing code with a severity level of EventMustBeAborted which should cause the event to be aborted in the run manager.  I assume then that the AnalyzeEvent() method of the persistency manager will not be called which is where the event is written out.  If it is called even when the event is aborted, we could check the state of the run manager to see if the event is being aborted.

I don't think we want to skip writing events based on whether some hits collection is filled or not.  It could be empty for some other reason than the biasing plugin.

JeremyMcCormick, 2016-11-10:
Is this still a problem with your photonuclear biasing class?  

I see you call AbortEvent() from within your code and I believe if this is working correctly then the end of event IO should not be activated.  Is that what you're seeing?

omar-moreno, 2016-11-10:
No, even though I call AbortEvent, the event gets written out.  I'm going to look into this now.

JeremyMcCormick, 2016-11-10:
Okay, probably the persistency manager needs to check in
AnalyzeEvent() that the run manager isn't aborting the event.  That would
be easy to add.

I know there is a hard abort for run called with AbortRun(false) but I
didn't see a similar option on the AbortEvent() method.

omar-moreno, 2016-11-10:
I'm testing something now.  I'll commit it if it works.

JeremyMcCormick, 2016-11-10:
Also, we should also make sure that aborting the event does not cause a memory leak because an object is not cleared or deleted properly.

PR #57 (2016-11-10): When biasing, only events which are fully processed should be written out.
This pull request resolves #49.
closed 2016-11-10 by JeremyMcCormick