LDMX-Software/ldmx-sw #50: RootPersistencyManager::Store message frequency

omar-moreno, 2016-11-10:
A user should be able to disable or set how often the "saving event" message is printed.  Currently, it's printed every event.

The frequency should be settable through the user provided macro.

omar-moreno, 2016-11-10:
This applies to the "Begin Event" and "End Event" should messages as well.

JeremyMcCormick, 2016-11-10:
I'm going to configure print outs from the ROOT IO based on the setting of the verbose level setting of the persistency manager.  Level 0 will have no output and Level 4 will print a lot including the output hit and particle objects.

As for event and run messages, right now they are hard-coded, but I can add macro commands to configure them.  

Would it be useful to have the event print outs in a sim plugin that can be enabled/disabled so I don't have to hard-code anything into the user actions?  I could add a G4Messenger for this plugin when it is created if you want to specify the "modulus" via a macro command (I think it should work).

omar-moreno, 2016-11-10:
I think adding it as a user plugin is fine.

JeremyMcCormick, 2016-11-10:
After the recent ROOT IO update you can set the verbose level of the ROOT persistency manager in Geant4 using a macro command like `/ldmx/persistency/root/verbose 4`.  Verbose level of 1 should have no output at all from the persistency manager and 4 will have a lot of low level debug print outs.

I am not planning on having a modulus that would only print every Nth event for the ROOT persistency manager.  If you don't want to see print outs then you should just use the default verbose setting of 1.  A verbose level of 2 will give you a small amount of output that just prints out when files are opened/closed and will print the collection sizes for every event.  The levels 3 and 4 can be used for debugging as they print some of the ROOT tree information and level 4 will print out all the collection objects.

For the individual event print outs, which is implemented separately outside of the persistency manager, I can give this a modulus parameter that would specify how often you want this information to print.

Is that good enough?

omar-moreno, 2016-11-10:
That should be good enough for now.

JeremyMcCormick, 2016-11-10:
Alright I'll close this issue.  There's a separate one for the event print outs.

closed 2016-11-10 by JeremyMcCormick