LDMX-Software/ldmx-sw #62: SimParticle methods getDaughters and getParents return null when using python.

omar-moreno, 2016-11-11:
Using SimParticle::geDaughters and SimParticle::getParents returns null when trying to access these methods through the python interface.  This is likely due to a std::vector being constructed on the fly.  Tthe methods work when they are changed to return a TRefArray directly.

These methods should be changed to either return the TRefArray to return a pre-filled std::vector.

JeremyMcCormick, 2016-11-14:
I guess the vector I am creating and returning in those methods is not being passed properly to Python.  

The two most obvious solutions would be returning the TRefArray so you could get a TRef entry and cast GetObject() to a SimParticle pointer.

The other option is adding getDaughterCount() and getMotherCount() methods along with getDaughter(int i) and getMother(int i) so you can access them directly.

Is either of these preferable?

I had hesitated to give access to the underlying TRefArray as it is a pointer and then it would be possible for the user to add or delete particles from this data structure.  But maybe we just need to trust people not to do something that stupid.  :)

omar-moreno, 2016-11-14:
I'm not sure what's best, but I don't think you need to worry about a user
adding and deleting particles from the data structure.  However, you
probably do want to hide the underlying data structure from users so adding
getters might be the way to go.  One thing to look at:  When iterating over
a TRefArray, is one way significantly faster than the other?

JeremyMcCormick, 2016-11-14:
It should be fixed now with PR #63.

closed 2016-11-14 by JeremyMcCormick