LDMX-Software/ldmx-sw #72: Support Mac/OSX build.

JeremyMcCormick, 2016-11-30:
Check that the software builds and runs successfully on OSX.

One needed change is that the default lib of SimPlugins is hardcoded to "libSimPlugins.so" which would be a .dylib instead on the Mac.  This can be handled with a C++ macro.

JeremyMcCormick, 2016-12-01:
[https://root.cern.ch/phpBB3/viewtopic.php?f=3&t=22778](https://root.cern.ch/phpBB3/viewtopic.php?f=3&t=22778)

JeremyMcCormick, 2017-02-02:
ROOT build seems okay now with these deps:

- ROOT 6.08.00

- OSX 10.12.1

- Xcode 8.1

- cmake version 3.7.0

- gcc Apple LLVM version 8.0.0 (clang-800.0.42.1)

Run CMake using

```
cmake ../root-6.08.00 -Dfortran=OFF
```

Then 'make -j4' seems to work okay and builds ROOT into the install dir.

JeremyMcCormick, 2017-02-02:
Working Geant4 CMake config:

```
cmake -DGEANT4_USE_GDML=ON -DGEANT4_INSTALL_DATA=ON -DGEANT4_USE_OPENGL_X11=ON -DXERCESC_LIBRARY=/work/ldmx/dev/xerces-c-3.1.4/lib/libxerces-c-3.1.dylib -DXERCESC_INCLUDE_DIR=/work/ldmx/dev/xerces-c-3.1.4/include/ -DCMAKE_INSTALL_PREFIX=/work/ldmx/dev/geant4.10.02.p02-install ..
```

JeremyMcCormick, 2017-02-02:
Xerces build error:

```
[63 $] ./bin/ldmx-sim
dyld: Library not loaded: libcurl.4.dylib
  Referenced from: /work/ldmx/dev/xerces-c-3.1.4/lib/libxerces-c-3.1.dylib
  Reason: Incompatible library version: libxerces-c-3.1.dylib requires version 8.0.0 or later, but libcurl.4.dylib provides version 7.0.0
Abort trap: 6
```

JeremyMcCormick, 2017-02-02:
https://cmake.org/Wiki/CMake_RPATH_handling

JeremyMcCormick, 2017-02-02:
Set in env to print library names as they are loaded:

```
DYLD_PRINT_LIBRARIES=1
```

JeremyMcCormick, 2017-02-03:
Yet another ROOT forum thread:

https://root.cern.ch/phpBB3/viewtopic.php?f=3&t=23099

omar-moreno, 2018-01-10:
Closing for now.  Will re-open when this becomes an issue.

closed 2018-01-10 by omar-moreno