LDMX-Software/ldmx-sw #82: Create v1 detector with latest Ecal/Hcal and LYSO target

omar-moreno, 2016-12-02:


omar-moreno, 2016-12-02:
The target should be assume to be made of Saint-Gobain LYSO which has 10% yttrium content. See 
[paper](http://authors.library.caltech.edu/8520/1/CHEieeetns07b.pdf) for the details.

omar-moreno, 2016-12-02:
For reference, Saint-Gobain LYSO has been defined as follows in the GDML: 

    <material formula="L1.8Y0.2Si05" name="Saint_Gobain LYSO" >
        <D value="7.4" unit="g/cm3"   />
        <fraction n="0.7145" ref="Lu" />
        <fraction n="0.0403" ref="Y"  />
        <fraction n="0.0637" ref="Si" />
        <fraction n="0.1815" ref="O"  />
    </material>

omar-moreno, 2016-12-03:
After some feedback from David Hitlin, the target should be assume to have 7% yttrium content.  The material is now defined as:
    <material formula="L1.86Y0.14Si05" name="LYSO" >
        <D value="7.4" unit="g/cm3"   />
        <fraction n="0.7297" ref="Lu" />
        <fraction n="0.0279" ref="Y"  />
        <fraction n="0.0630" ref="Si" />
        <fraction n="0.1794" ref="O"  />
    </material>

See [paper]( http://www.hep.caltech.edu/~zhu/papers/04_tns_lyso.pdf ) for the details.

JeremyMcCormick, 2016-12-08:
Resolved by PR #86

omar-moreno, 2016-12-08:
A scintillator needs to be added upstream of the target.

JeremyMcCormick, 2016-12-13:
This is basically resolved by #87.

closed 2016-12-13 by JeremyMcCormick