LDMX-Software/ldmx-sw #98: EcalVetoProcessor segfaults

omar-moreno, 2016-12-15:


ocolegro, 2016-12-15:
Hi, I believe this issue is because I forgot to set the # of layers to be appropriate for the v0 detector, I can fix this by changing the default number of layers to something large, such as 100.

omar-moreno, 2016-12-15:
How many layers are in v1? Also, can we get these numbers from the geometry
itself?

ocolegro, 2016-12-15:
40 layers for v1.  I found segfaults as well in earlier testing, but they went away when I changed the preset number of layers to something larger.  I forgot to fix this on my PR.

I'm not sure how to get the number of layers from the FWK.

omar-moreno, 2016-12-15:
Alright, let me fix it since I'm actively working on it. I'll let you know
if it works.

JeremyMcCormick, 2016-12-15:
You can't currently get geometry information easily from the EventProc classes.  This is something that is on my list...

omar-moreno, 2016-12-15:
Changing the number of layers to 40 fixes one of the seg faults.  I see another one being caused by the unpacking of a specific detector ID: 

===========================================================
There was a crash.
This is the entire stack trace of all threads:
===========================================================
#0  0x00007f6eb7eb06bc in waitpid () from /lib64/libc.so.6
#1  0x00007f6eb7e3a80c in do_system () from /lib64/libc.so.6
#2  0x00007f6ebc26e249 in TUnixSystem::StackTrace (this=0x15df8e0) at /home/omoreno/work/physics_packages/root/core/unix/src/TUnixSystem.cxx:2324
#3  0x00007f6ebc26ff3c in TUnixSystem::DispatchSignals (this=0x15df8e0, sig=kSigSegmentationViolation) at /home/omoreno/work/physics_packages/root/core/unix/src/TUnixSystem.cxx:3562
#4  <signal handler called>
#5  0x00007f6eb8bf7911 in detdescr::DetectorID::unpack() () from /home/omoreno/work/ldmx/software/ldmx-sw/install/lib/libDetDescr.so
#6  0x00007f6eb904483b in eventproc::EcalVetoProcessor::hitToPair(event::SimCalorimeterHit*) () from /home/omoreno/work/ldmx/software/ldmx-sw/install/lib/libEventProc.so
#7  0x00007f6eb9042c02 in eventproc::EcalVetoProcessor::execute() () from /home/omoreno/work/ldmx/software/ldmx-sw/install/lib/libEventProc.so
#8  0x00007f6eb904800d in eventproc::EventLoop::run(int) () from /home/omoreno/work/ldmx/software/ldmx-sw/install/lib/libEventProc.so

The ID is always the same and the strange thing is that it happens randomly.  I'm not sure what this is all about ...

JeremyMcCormick, 2016-12-19:
What do you mean the seg fault happens randomly?  Does that mean it sometimes works on the same ID on which it sometimes crashes?

That is indeed quite strange ...

What is the specific ID which causes the crash?  It is possible that I can make some kind of test which would reproduce this.

JeremyMcCormick, 2016-12-20:
Also, can you compile with `CMAKE_BUILD_TYPE=Debug` so that line numbers show up in your traceback?  It is hard to say what is going wrong unless I have the exact line number where the crash occurs.

JeremyMcCormick, 2017-01-17:
Should we just close this since this class is being replaced/rewritten?

omar-moreno, 2017-01-17:
Yes, that's fine.

omar-moreno, 2017-02-18:
I'm reopening this issue because the seg fault is still present.  I'll add more info soon.

omar-moreno, 2017-02-24:
This is resolved (again) by commit f340c23008d8d09993acc7e64c563d2395445651.

closed 2017-02-24 by omar-moreno