LDMX-Software/ldmx-sw #145: Make producer collection names configurable.

omar-moreno, 2017-01-25:


JeremyMcCormick, 2017-01-26:
Isn't this just a matter of reading string parameters in the Producer's configure() method and using them as the collection names?

So I think this is up to each individual Producer to implement.

omar-moreno, 2017-02-10:
Yes, but currently the names are hard coded into each producer.  I'm thinking they should be configurable ..

JeremyMcCormick, 2017-03-21:
> Yes, but currently the names are hard coded into each producer. I'm thinking they should be configurable ..

I think it is up to each individual Producer to implement this behavior with a string parameter, if it makes sense.  That's just a few lines of code, e.g. to read "input_collection" or "output_collection" as a string param in the configure() method.

But given that downstream Producer or Analyzer classes may depend on collections being there with a certain name, I'm not sure that every collection name should actually be overridable by the user, especially for important ones like digit collections created from sim hit input, etc.

It probably makes a lot more sense where you might want to have multiple collections from the same Producer classes run within a job, like for instance sets of Tracks made from different cuts.

closed 2018-01-10 by omar-moreno