LDMX-Software/ldmx-sw #146: Rewrite ECal geometry in GDML detector description

JeremyMcCormick, 2017-01-25:
The current ECal geometry description is not structured very well -- it is essentially using thousands of polyhedral planes, as dumped from the Cal Geant4 runtime.

Instead, each ECal layer can be defined as the union of two trapezoids.  

Each module should have its own logical volume which can be assigned a module number.

The layer structure should be built using GDML loops.

This should make the detector description much easier to read as well as making it more suitable for changes to the layering structure, which are difficult to implement now without dumping GDML from the Calo sim.

ocolegro, 2017-04-13:
Hi Guys, slides 10 and 11 below contain all relevant information about the longitudinal profile for the current ECAL.

Basically the ECAL starts out with an Si layer and no absorber, followed by a slow ramp in absorber thickness, until a constant thickness of ~1 X0 is reached.

Thick layers are then placed in the very back in order to catch the tail end of E&M fluctuations.

[LDMX_UCSBNewGeom_11_28_2016.pdf](https://github.com/LDMXAnalysis/ldmx-sw/files/920816/LDMX_UCSBNewGeom_11_28_2016.pdf)

nhanvtran, 2017-06-13:
Here's a first visual look at the two geometries...

old (v2)
[g4_00-v2.pdf](https://github.com/LDMXAnalysis/ldmx-sw/files/1072743/g4_00-v2.pdf)

new (v3)
[g4_01-v3.pdf](https://github.com/LDMXAnalysis/ldmx-sw/files/1072744/g4_01-v3.pdf)

looks reasonable by eye at least.  will try to make some plots now.

nhanvtran, 2017-06-13:
Here's just sum of energy deposited in the silicon for

old (v2)
[image]

new (v3)
[image]

also looks reasonable

JeremyMcCormick, 2017-06-13:
Looks good!

Can you generate 10x more statistics easily?

nhanvtran, 2017-06-13:
10x more stats:

old (v2)
[image]

new (v3)
[image]

there are some outliers, not sure why...but it's more consistent than with less stats.

nhanvtran, 2017-06-14:
@awhitbeck @omar-moreno @JeremyMcCormick 

Ok, so now we can start to see where the difference is... 
Black (v2, old)
Red (v3, new)

energies of each hit
[image]
energies of each hit, log scale
[image]
energy per hit per event as a function of z position
[image]
sum energy per event
[image]

nhanvtran, 2017-06-27:
Ok, I've been staring at this for a while now...the only difference I can find between the 2 geometries is actually in the pre shower...
The V2 geometry has an extra 1mm of Al in the front of it, otherwise, they are identical.  See the text comparison below.  

Now why this translates into a different shower profile is a bit confusing to me (unless I missed something and they are in fact different). 

[image]

blue is new, black is old
[image]

omar-moreno, 2017-07-25:
This issue was resolved by PR 44b98c385ec81d649d2ccb589610b98b6dcc2fdc.

closed 2017-07-25 by omar-moreno