LDMX-Software/ldmx-sw #149: Indent source code within namespace

JeremyMcCormick, 2017-01-28:
There should be a 4 space indent within the ldmx namespace.

omar-moreno, 2017-01-30:
What made you change your mind on this?  You were very adamant about this
... ​

JeremyMcCormick, 2017-01-30:
It was pointed out to me rather convincingly that it is bad C++ style to not match bracket/indentation scope so I think we should change it.  It is helpful to see that you are in a namespace within classes and functions from the indentation and I think indenting is the standard style.

I wanted to make sure it gets a proper issue as it could potentially make merging in branches quite a pain so I we should discuss about when it would be best to do it.  Probably I would try to do it after the biasing branch gets merged.

omar-moreno, 2017-01-30:
OK good, I'm glad someone else convinced you because no matter how much I
tried, I couldn't :).

PR #263 (2017-04-05): Cleanup indentation and code style
- Indented all code within `ldmx` namespace.

- Also includes some minor code style cleanup.

- Resolves #149.
closed 2017-04-05 by JeremyMcCormick