LDMX-Software/ldmx-sw #150: Make EcalVeto fully configurable

awhitbeck, 2017-01-30:
All parameters for ecalveto are currently hardcoded.  We should make them configurable through configuration files.  

There is also a mistake in this code, I think.  Memory for three cut variables is being set, but all three variables are push into the first element of varaibles_:

https://github.com/LDMXAnalysis/ldmx-sw/blob/master/EventProc/src/EcalVetoProcessor.cxx#L108-L110

awhitbeck, 2017-01-30:
A first attempt to address this issue has been commit to iss150

omar-moreno, 2017-02-10:
@awhitbeck I'm going to add a class to encapsulate the Ecal veto results so we don't need to rely on using the TriggerResults class.

Also, @ocolegro said he made changes to the Ecal veto algorithm.  We should just port those changes as part of this issue.

omar-moreno, 2017-02-18:
Before I fix the EcalVetoProcessor, I need the changes merged in.

closed 2017-02-18 by omar-moreno