LDMX-Software/ldmx-sw #187: Problems closing input file when running Recon

nhanvtran, 2017-02-23:
This depends on on the order of the recon modules...

JeremyMcCormick, 2017-02-24:
Needs more info I think.  How to reproduce?

nhanvtran, 2017-02-24:
So this works, but crashes if you remove the `findable_track` module:
```
import sys
from LDMX.Framework import ldmxcfg
p=ldmxcfg.Process("recon")
p.libraries.append("libEventProc.so")
ecalDigis = ldmxcfg.Producer("ecalDigis","ldmx::EcalDigiProducer")
ecalDigis.parameters["meanNoise"] = 0.015
ecalDigis.parameters["readoutThreshold"] = ecalDigis.parameters["meanNoise"]*3
hcalDigis = ldmxcfg.Producer("hcalDigis", "ldmx::HcalDigiProducer")
hcalDigis.parameters["meanNoise"] = 1.5
ecalVeto = ldmxcfg.Producer("ecalVeto", "ldmx::EcalVetoProcessor")
ecalVeto.parameters["num_ecal_layers"] = 33
ecalVeto.parameters["back_ecal_starting_layers"] = 20
ecalVeto.parameters["num_layers_for_med_cal"] = 10
ecalVeto.parameters["total_dep_cut"] = 25.
ecalVeto.parameters["total_iso_cut"] = 15.
ecalVeto.parameters["back_ecal_cut"] = 1.
ecalVeto.parameters["ratio_cut"] = 10.
trigger = ldmxcfg.Producer("trigger", "ldmx::TriggerProcessor")
trigger.parameters["threshold"] = 12.0
trigger.parameters["mode"] = 0
trigger.parameters["start_layer"] = 1
trigger.parameters["end_layer"] = 16
findable_track = ldmxcfg.Producer("findable", "ldmx::FindableTrackProcessor")
p.sequence=[ecalDigis, hcalDigis, ecalVeto, trigger, findable_track]
p.inputFiles=["/u/br/omoreno/ldmx/data/mc/4pt0_gev_e_target_gamMuConv_v1_fieldmap/ldmx_sim_events_gun_job0_energy4.00_th0.00_gamMuTargetSim.root"]
p.outputFiles=["ldmx_sim_events_gun_job0_energy4.00_th0.00_gamMuTargetSim_recon.root"]
p.printMe()
```

omar-moreno, 2018-01-10:
I haven't been able to reproduce this using the latest recon.  Maybe there was an issue with our old setup? Not sure, but closing this for now.

closed 2018-01-10 by omar-moreno