LDMX-Software/ldmx-sw #190: Create v2 detector with larger recoil tracker axial layer and Hcal with 10 mm absorber.

omar-moreno, 2017-02-24:


JeremyMcCormick, 2017-02-24:
[This commit](https://github.com/LDMXAnalysis/ldmx-sw/commit/252ff334b7729c044baa72ac2fe8b0abb091f631) adds the changes to recoil layer 5.

awhitbeck, 2017-02-26:
I have changed the hcal absorber from 50mm to 10 mm and tripled the number of layers so that the total interaction length was constant.   Geometry before:

[image]

Geometry after changes:

[image]

omar-moreno, 2017-03-01:
This issue is resolved by PR 5d861062ed838b8883ae191a78a8bc603af88145

closed 2017-03-01 by omar-moreno