LDMX-Software/ldmx-sw #239: Get GEANT3 to work with ldmx-sw

nhanvtran, 2017-03-27:


JeremyMcCormick, 2017-03-28:
Not sure this is something that belongs in our issue tracker until we know what we want to accomplish with this and have some actual plan on how to do it.  Otherwise, it is just likely to sit here for months without any action.

Also, if the PN modeling is supposedly better in GEANT3 then I would like to see some evidence from the literature that this is the case before we go down this rabbit hole.

omar-moreno, 2017-03-28:
Is there evidence in the literature that Geant4 is better? ;)

JeremyMcCormick, 2017-04-04:
There are no concrete plans to do this and the technical hurdles are non-trivial.  Closing for now.  (Re-open in the unlikely chance you are actually going to work on it!)

closed 2017-04-04 by JeremyMcCormick