LDMX-Software/ldmx-sw #243: Framework classes should include header dependencies

JeremyMcCormick, 2017-03-28:
For instance, EventProcessor has these forward class declarations:

```
class Event;
class ParameterSet;
class Process;
class EventProcessor;
class RunHeader;
```

Every EventProcessor implementation then needs to include the headers for these classes in its cxx file, and if not cryptic compilation errors occur.

Instead, these headers should simply be included in EventProcessor, given that they are actual dependencies of this class and required by all sub-classes.

jmmans, 2017-03-28:
I do not agree in most of these cases.  I can agree that Event is a true dependency, but if an EventProcessor does not want to access the RunHeader (most won't want to) or use the ParameterSet (some will not not need to), then they need not suffer from the header pollution.

JeremyMcCormick, 2017-03-28:
It doesn't matter if the user's EventProcessor class uses these dependencies.  They *have* to be included someplace -- otherwise, their code will not compile.  I'm not really seeing how it is an advantage to anyone if including "EventProcessor.h" does not actually also include all the headers it actually depends on and needs in order to build without errors.  If the user forgets to include these disguised dependencies, then they end up with cryptic compile/linking errors having to do with the classes not being defined.  And then every Producer or Analyzer class has to include this same set of headers for the classes you have forward declared.  Disguising transitive dependencies like this is going to be very confusing for people, especially for anyone who is not that familiar with C++.

And "header pollution" I believe is a non-issue, especially in a code base of ours which is of fairly limited size.  There are libraries like boost that are almost completely defined in header files, and it causes no real issues at all.  The inclusion of the header code is handled perfectly fine by the include guards.  (Please correct me if I'm wrong and explain why header pollution is something we actually need to worry about.)

jmmans, 2017-03-28:
I will need to look at examples.  Process should be directly used by _no_one_. Every use of "Process" in the header is a reference, so it should never be required -- that's an "internal-implementation-detail".  Are you getting a case where the Process is actually required to be included?

jmmans, 2017-03-28:
Anyway, I've created PR 246 to meet this request.

closed 2017-03-28 by jmmans

JeremyMcCormick, 2017-03-28:
Great,thanks!