LDMX-Software/ldmx-sw #251: Generic tracking enhancement issue

JeremyMcCormick, 2017-03-30:
Placeholder issue for adding tracking to the framework.

Ideas and requirements:

- Implementation of charge deposition simulation (based on sisim from hps-java)

- Simple track following based on algorithm similar to seed tracker (seeding with extension to additional layers using helix swimming)

- Cellular automata for initial track finding

- Deterministic Annealing Filter (DAF) for track fitting

Questions:

- Can we use GENFIT as a starting point?

- Can we use any code in cmssw?  (includes both cellular automata and DAF code)

Actual branch(es) don't need to be made based on this issue but it can be used as a place for discussion.

omar-moreno, 2017-03-30:
Ummmm ... These should all be their own issue which I will make when I get
to them.  This issue is pointless and should be closed.

It's odd to make a generic tracking issue ...

JeremyMcCormick, 2017-03-30:
This is a meta-issue to discuss what needs to be turned into an actual issue.  :)

omar-moreno, 2017-03-30:
... isn't that what Slack is for?

JeremyMcCormick, 2017-03-30:
Well, stuff disappears on Slack.  I just figured we wanted a place to put links, ideas, etc.  This seems like a good enough place as any.  I think probably what we should do is have a tracking "project" and put a bunch of issues under it.

JeremyMcCormick, 2017-04-04:
May depend on #171 being merged.

omar-moreno, 2017-05-16:
This isn't needed so I'm closing it.

closed 2017-05-16 by omar-moreno