LDMX-Software/ldmx-sw #252: add material for magnet to geometry

awhitbeck, 2017-03-30:
For understanding the effect of signal inefficiencies due to electrons hitting the side HCal, we need to first include material from magnet which might cover this wide angle region.  Need to know roughly what the material is and the dimensions.

omar-moreno, 2017-06-05:
I received a drawing of the magnet used for HPS.  I modified the size of the shims such that the gap is 14" instead of the 8" that HPS uses. A drawing of the resulting magnet is attached.
[ldmx_magnet.pdf](https://github.com/LDMXAnalysis/ldmx-sw/files/1053126/ldmx_magnet.pdf)

omar-moreno, 2017-06-07:
Screenshot of magnet loaded into GDML.  At some point, this should be rewritten so tessellated volumes aren't used for the iron blocks. 

[image]

omar-moreno, 2017-06-07:
For future reference, I used https://www.solveering.com/InStep/instep.aspx to convert from an STL file directly to GDML. I believe InStep also allows you to convert from GDML --> CAD @awhitbeck.

omar-moreno, 2017-07-17:
Moving away from using tessellated solids to describe the iron blocks seemed to have solved the following issue: 
```
-------- WWWW ------- G4Exception-START -------- WWWW -------
*** G4Exception : GeomNav1002
      issued by : G4Navigator::ComputeStep()
Track stuck or not moving.
          Track stuck, not moving for 10 steps
          in volume -ldmx_magnet_PV- at point (-574.699,177.8,-721.64) (local point (-574.699,177.8,-321.64))
          direction: (-0.0458547,0.658799,-0.75092) (local direction: (-0.0458547,0.658799,-0.75092)).
          Potential geometry or navigation problem !
          Trying pushing it of 1e-07 mm ...Potential overlap in geometry!

*** This is just a warning message. ***
-------- WWWW -------- G4Exception-END --------- WWWW -------
```

omar-moreno, 2017-07-17:
New magnet geometry using G4 solids to describe the iron blocks
[image]

omar-moreno, 2017-07-17:
Running 1000 4 GeV electrons on target with the magnet material in:

real	10m13.133s
user	9m48.037s
sys	0m0.418s

omar-moreno, 2017-07-17:
Without the magnet material, it's about 45 seconds faster:

omar-moreno, 2017-07-25:
This issue was resolved by PR 44b98c385ec81d649d2ccb589610b98b6dcc2fdc.

closed 2017-07-25 by omar-moreno