LDMX-Software/ldmx-sw #260: EventFile should allow appending to an existing file

JeremyMcCormick, 2017-04-05:
Especially useful for the sim app.

jmmans, 2017-04-14:
How do we handle the case if the set of tree branches isn't the same when you re-open?  The job doesn't have to be configured the same way that the original job was.  Also, often it's dangerous/bad practice to be able to re-write to the same file...

JeremyMcCormick, 2017-04-14:
The main use case for this is the simulation, where you may have multiple Geant4 runs in one session, in which case you will likely want to append events to the same file rather than create a new file or truncate the existing file.

I would just assume that the user is going to add events with the same tree structure as exists in the file.  I don't think you would need to check for this specifically, and if someone does it then assume they are shooting themselves in the foot and it is their problem.  :)

closed 2018-07-20 by omar-moreno