LDMX-Software/ldmx-sw #268: Add beam event generator to simulation

JeremyMcCormick, 2017-04-13:
- Beam spread

- Bunches with multiple electrons

jmmans, 2017-04-14:
As part of this, I would suggest that we should have a "beam spot" class which can be configured to set up the beam spot for use in multiple parts of the code (e.g. for integrating Madgraph events as well just inserting particle-gun-based additional beam electrons)

JeremyMcCormick, 2017-05-03:
Hi, @nhanvtran.

I saw your commits on this issue's branch but this is not the direction I was thinking of going for this feature.

There is already a generic multi-particle source provided by Geant4:

https://geant4.web.cern.ch/geant4/UserDocumentation/UsersGuides/ForApplicationDeveloper/html/ch02s07.html

Instead what I was imagining here is a primary generator that takes beam parameters or provides them as a default.

For instance, in the DASEL design report the beam spot is described as a 4 x 4 cm area, so you could provide a random XY position within this region.

We may also want to specify a time based on the assumption of 10^8 electrons / s also described in the DASEL proposal.

If we want to arbitrarily add multiple particles with more complex parameters such as a randomized region for the origin, the GPS already provides this so we can just make that generator available within the sim app.

nhanvtran, 2017-05-03:
Hey @JeremyMcCormick 

Thanks, I committed something fast so I could send something to Melissa (who's having a look), so it's obviously very incomplete, but at least let me say the rationale

I was planning on merging two tasks into one:
1. pileup
2. beam spread

For the pileup, the issue with the GPS is that I don't know how to generate a Poisson-varied number of guns per event.  It didn't look possible from the documentation (you also pointed to it).  So I was writing a custom generator.  

Then for the beam spread, I haven't yet written anything on that but I was going to add a method in the PrimaryGeneratorAction.cxx which then spread the beam no matter the primary generator (LHE, ROOT, MultiParticleGun,ParticleGun).  I didn't want to write a separate new primary generator for the beam spot since it will act the same on all primary generators.  

At least that was the plan...

omar-moreno, 2017-05-08:
@nhanvtran @JeremyMcCormick Just want to make sure that smearing of vertices when reading in LHE files is still needed even after implementing this generator?  If so, that's something I can quickly implement.

nhanvtran, 2017-05-08:
Hi @omar-moreno, sorry I got pulled away from this last week due to some travel, but I hope to get a first working version in the next day or so.  The plan is to try to have some generic beam smearing no matter the primary generator.  If that sounds okay, let me try to put something in first...I'll keep you posted.

JeremyMcCormick, 2017-05-08:
I think that the vertex smearing should be implemented as a sim plugin to have the most flexibility.

nhanvtran, 2017-05-09:
@JeremyMcCormick I guess this is not what you wanted (I just saw the message above), but let me know what you think.  The relevant code for the smearing is here:
https://github.com/LDMXAnalysis/ldmx-sw/blob/iss268/SimApplication/src/PrimaryGeneratorAction.cxx#L53-L72

To enable the beamspot smearing, you do, for example:
```
/ldmx/generators/beamspot/enable
/ldmx/generators/beamspot/size 20.0
```

It's pretty straight-forward, so I don't know if you still wanted the standalone plugin.  Otherwise, the rest of the branch has a skeleton for producing pileup.  I can split that off if you want.

JeremyMcCormick, 2017-05-09:
This looks good overall so far.

I have one minor change I would like to see here:

```
if (PrimaryGeneratorMessenger::useBeamspot()) smearingBeamspot(event);
```

Instead it would be preferable here that a boolean flag is set on PrimaryGeneratorAction when beamspot smearing is enabled and then it would just turn into:

```
if (useBeamspot_) smearingBeamspot(event);
```

Static access as you have coded it is to be avoided if possible and the messenger classes are not really intended to carry around application state unless it is absolutely necessary.  (I know there is a lot of similar code in the biasing but I do not want to add any more of it to the codebase.)

JeremyMcCormick, 2017-07-10:
This is resolved by addition of the mult-particle generator.

closed 2017-07-10 by JeremyMcCormick