LDMX-Software/ldmx-sw #304: Add DetID module

JeremyMcCormick, 2017-07-13:
The following classes can be moved out of DetDescr into a DetID module:

DefaultDetectorID
DetectorID
DetectorIDStore
EcalDetectorID
IDField

These classes have no dependence on geometry classes, so it would be better to establish a dependency to them from DetDescr.

tomeichlersmith, 2019-06-28:
Possible solution on pull request #564 .

omar-moreno, 2019-07-03:
I'm not sure what the original idea was here.  Maybe @tomeichlersmith has some insight?

tomeichlersmith, 2019-07-03:
I believe the original idea was to make it easier to find the ID-related classes by separating them from the geometry description related classes. This separation would also means the DetDescr module wouldn't get too bloated as more geometry/ID classes were created. It is not a necessary change for any other development (to my knowledge), so the issue could be closed without merging if it would just lead to confusion.

omar-moreno, 2019-07-03:
Not against it merging it.  We actually have a lot of instances where separating classes out would be cleaner.  However, I have been hesitant because this would also lead to having way to many modules.  Let me think about this a bit.