LDMX-Software/ldmx-sw #321: Add directory for GammaMuMu analysis

awhitbeck, 2017-07-25:
Seems time to start putting analysis code into ldmx-sw:

- Write up analyzers for plotting all of the validation variables for the gammaMuMu studies 
 - q-squares var
 - muon kinematic vars laid out [here](https://ldmxsoftware.slack.com/files/nataliatoro/F6AJTS5RB/pseudo-code_for_some_of_the_muon_kinematic_variables_used_by_geant_in_muon_pair_generation.txt)
- Put in code `GammaMuMuAnalysis`

nhanvtran, 2017-07-26:
i thought we would use this repository for analysis code? 
https://github.com/LDMXAnalysis/ldmx-sw-scripts

this admittedly needs to be cleaned up, but could be used

awhitbeck, 2017-07-26:
well, I am willing to put things other places, but I would like to use the framework so that configurations for a corresponding analysis can be built up.

omar-moreno, 2017-07-26:
I'm also guilty of using a different repo to store things: 
https://github.com/LDMXAnalysis/ldmxpy

But I agree with @awhitbeck we should start building up an analysis chain that we can start running over all samples.

nhanvtran, 2017-07-26:
not sure what @JeremyMcCormick will say, but i think we should keep it separate from the core code-base.  but we should start collecting and documenting the analysis side.  maybe a new repo `ldmx-analysis` or something like that

omar-moreno, 2017-07-26:
I'm fine with either.  Let's talk on Slack.

omar-moreno, 2018-02-16:
We decided against putting analysis code in ldmx-sw, so I'll close this issue.

closed 2018-02-16 by omar-moreno