LDMX-Software/ldmx-sw #339: Fix Ecal top and bottom module edges when gaps are enabled

ASPatterson, 2017-08-04:
With gap sizes of 2mm (but not 0 or 1 mm), EcalHexReadout receives requests for cellModuleIDs for positions which should be outside of the ecal geometry. Specifically, along the top edge of module 1 (top) and the bottom edge of module 4 (bottom), between max abs y and 2 mm above it.  

With a single-particle gun at (46.25, 259.03, -.55), we see EcalScoringPlaneHits in front but not in back of the Ecal. There are of course no EcalSimHits above the expected max(y) cell at 254.5. The problem lies in the ecal.gdml geometry.  

```
root [3] LDMX_Events->Scan("eventNumber_:EcalScoringPlaneHits_sim.x_:EcalScoringPlaneHits_sim.y_","EcalScoringPlaneHits_sim.layerID_==1 && EcalScoringPlaneHits_sim.y_ > 257 && EcalScoringPlaneHits_sim.y_<260 && abs(EcalScoringPlaneHits_sim.x_)<40","colsize=25");
***********************************************************************************************************
*    Row   * Instance *              eventNumber_ * EcalScoringPlaneHits_sim. * EcalScoringPlaneHits_sim. *
***********************************************************************************************************
*       75 *       89 *                        77 *      -8.15025901794433594 *        257.87384033203125 *
*      237 *       53 *                       240 *       -5.0576176643371582 *        259.99383544921875 *
*      370 *       15 *                       375 *      -37.2833747863769531 *          258.872802734375 *
*      395 *        1 *                       400 *      -8.72008132934570312 *          258.324951171875 *
*      433 *      348 *                       439 *      -13.4030752182006836 *       259.033233642578125 *
*      439 *      117 *                       445 *      -35.4349861145019531 *       258.955230712890625 *
*      679 *       10 *                       687 *       9.45399093627929688 *         259.1495361328125 *
*      706 *      282 *                       714 *      -15.6735239028930664 *       259.127838134765625 *
*      759 *        8 *                       768 *       25.3161811828613281 *         257.7235107421875 *
*      771 *       23 *                       780 *      -14.8159990310668945 *         257.4217529296875 *
*      825 *      128 *                       835 *      -7.82471323013305664 *         258.2803955078125 *
*      911 *       92 *                       922 *      -15.9233741760253906 *       257.460052490234375 *
***********************************************************************************************************

root [2] LDMX_Events->Scan("eventNumber_:EcalScoringPlaneHits_sim.x_:EcalScoringPlaneHits_sim.y_","EcalScoringPlaneHits_sim.layerID_==2 && EcalScoringPlaneHits_sim.y_ > 257 && EcalScoringPlaneHits_sim.y_<260 && abs(EcalScoringPlaneHits_si***********************************************************************************************************
*    Row   * Instance *              eventNumber_ * EcalScoringPlaneHits_sim. * EcalScoringPlaneHits_sim. *
***********************************************************************************************************
*       12 *        1 *                        12 *       5.42727518081665039 *        259.09197998046875 *
*      104 *      168 *                       106 *       33.9969596862792969 *         257.9923095703125 *
*      164 *       42 *                       167 *      -15.2681989669799805 *         259.5032958984375 *
*      209 *        9 *                       212 *       -37.186737060546875 *            259.2626953125 *
*      234 *        4 *                       237 *       7.71273612976074219 *       259.683807373046875 *
*      335 *       61 *                       339 *      -19.7238941192626953 *       257.050994873046875 *
*      381 *      161 *                       386 *       18.4483566284179688 *       259.844940185546875 *
*      432 *      415 *                       438 *        38.307098388671875 *         257.7454833984375 *
*      433 *       63 *                       439 *       39.8167457580566406 *       257.653228759765625 *
*      477 *      498 *                       484 *        -36.37408447265625 *       259.533477783203125 *
*      655 *      207 *                       663 *      -18.5455398559570312 *        259.89886474609375 *
*      694 *       51 *                       702 *       19.3651275634765625 *       258.975433349609375 *
*      706 *      166 *                       714 *      -10.5742158889770508 *         259.7061767578125 *
*      706 *      193 *                       714 *        1.1655811071395874 *        258.98992919921875 *
*      706 *      274 *                       714 *      -17.6800975799560547 *       257.152557373046875 *
*      764 *      146 *                       773 *      -24.4676551818847656 *       258.523956298828125 *
*      778 *       11 *                       787 *       13.4527149200439453 *       258.330352783203125 *
*      799 *       15 *                       809 *       2.61872744560241699 *        257.60980224609375 *
*      799 *       74 *                       809 *       23.6866550445556641 *       259.983367919921875 *
*      825 *      330 *                       835 *      -20.8725166320800781 *        258.13861083984375 *
*      868 *      163 *                       879 *       25.9926414489746094 *       259.035797119140625 *
*      882 *       12 *                       893 *      0.739514589309692383 *        257.02215576171875 *
*      886 *      411 *                       897 *       -22.600250244140625 *            257.9716796875 *
*      924 *      118 *                       935 *      -34.7651901245117188 *       258.353240966796875 *
*      924 *      244 *                       935 *      -29.6718311309814453 *         258.6016845703125 *
Type <CR> to continue or q to quit ==> 
*      929 *        1 *                       940 *       33.4004096984863281 *        259.50701904296875 *
***********************************************************************************************************
```

ASPatterson, 2017-08-09:
[image]
[image]
Hcal hits before fix

ASPatterson, 2017-08-09:
[image]
[image]
Hcal hits for gap = 2 mm and below macro, before this fix.
```
# Parse the detector description using the GDML reader.  This assumes that
# there is a local soft link that points to the detector of interest.
/persistency/gdml/read detector.gdml

/ldmx/pw/enable
/ldmx/pw/read spdetector.gdml

/random/setSeeds 5 6
#2 3 on evt 2
# 28 29
# Enable biasing and set parameters.  This configuration specifies that
# the biasing operator is applied to gammas with energy above 2500 MeV
# that enter the Ecal.  Also, the photonuclear xsec is increased by a factor
# of 1000 within the Ecal only.
/ldmx/biasing/enable
/ldmx/biasing/particle gamma
/ldmx/biasing/process photonNuclear
/ldmx/biasing/volume ecal
/ldmx/biasing/threshold 2500

# Initialize the run
/run/initialize

/ldmx/biasing/xsec/particle gamma
/ldmx/biasing/xsec/process photonNuclear
/ldmx/biasing/xsec/threshold 2500
/ldmx/biasing/xsec/factor 1000

# Position the electron gun upstream of the target.  The initial energy
# of the electron will be 4 GeV.
#/gun/particle e-
#/gun/energy 4.0 GeV
#/gun/position 0 0 -0.55 mm
#/gun/position -12.458221 257.503826 -0.55 mm
#/gun/position 0 254.9 -0.55 mm
#/gun/position 3.628210 258.669837 -.55 mm
#/gun/direction 0 0 4 GeV

/ldmx/generators/mpgun/enable
/ldmx/generators/mpgun/nInteractions 1
/ldmx/generators/mpgun/pdgID 11
/ldmx/generators/mpgun/vertex 0 0 -0.55 mm
/ldmx/generators/mpgun/momentum 0 0 4 GeV

/ldmx/generators/beamspot/enable
/ldmx/generators/beamspot/sizeX 600
#; ; 20.0 #total extent 2 x 4 cm
/ldmx/generators/beamspot/sizeY 600
#40.0

# Plugin used to print an event status.
/ldmx/plugins/load EventPrintPlugin

# Plugin used to filter events with a hard brem.  In this case, the filter
# is being applied to the target.  Only events where the recoil electron
# has an energy below 1500 MeV are considered.  Also, only events that
# contain at least a single brem gamma above 2500 MeV are considered.
#/ldmx/plugins/load TargetBremFilter libBiasing.so
#/ldmx/plugins/TargetBremFilter/volume target_PV
#/ldmx/plugins/TargetBremFilter/recoil_threshold 1500
#/ldmx/plugins/TargetBremFilter/brem_threshold 2500
# Plugin used to filter events where a gamma undergoes a photo-nuclear
# reaction in the target.  In this case, the filter is being applied
# to the Ecal.
#/ldmx/plugins/load EcalProcessFilter libBiasing.so
#/ldmx/plugins/EcalProcessFilter/volume ecal

# Print a status every 1000 events.
/ldmx/plugins/EventPrintPlugin/modulus 1000

/ldmx/plugins/load TrackFilterPlugin
/ldmx/plugins/TrackFilterPlugin/verbose 0

#/ldmx/plugins/TrackFilterPlugin/process photonNuclear true
#/ldmx/plugins/TrackFilterPlugin/region CalorimeterRegion true
#/ldmx/plugins/TrackFilterPlugin/create pnFilter

#/ldmx/plugins/TrackFilterPlugin/region CalorimeterRegion true
#/ldmx/plugins/TrackFilterPlugin/create allFilter

# Write the events that pass the filter
/ldmx/persistency/root/verbose 2
/ldmx/persistency/root/file ldmx_sim_events.root

#/tracking/verbose 4

/run/beamOn 1000
```

ASPatterson, 2017-08-09:
[image]
[image]

And after applying the fix

ASPatterson, 2017-08-09:
Overlap check afterwards is clean.

omar-moreno, 2017-08-10:
This issue was resolved by PR 50907a4957229a94a4f11a93668de2601a9deb87.

closed 2017-08-10 by omar-moreno