LDMX-Software/ldmx-sw #340: Fix copynumber mistake

ASPatterson, 2017-08-06:
A copynumber was typed incorrectly, leading to rare hits being ID'd differently by Geant and by ldmxsw. Spot the problem:  
```
        <physvol name="Si_physvol_PS_center1" copynumber="0">
        <physvol name="Si_physvol_PS_center2" copynumber="7">
          <physvol name="Si_physvol_PS1" copynumber="h_ps">
          <physvol name="Si_physvol_PS2" copynumber="h_ps+7">
          <physvol name="Si_a_center_physvol1" copynumber="(a-1)*14+14">
          <physvol name="Si_a_center_physvol2" copynumber="(a-1)*14+21">
          <physvol name="Si_a_ring1_physvol1" copynumber="(a-1)*14+15">
          <physvol name="Si_a_ring1_physvol2" copynumber="(a-1)*14+22">
          <physvol name="Si_a_ring2_physvol1" copynumber="(a-1)*14+16">
          <physvol name="Si_a_ring2_physvol2" copynumber="(a-1)*14+23">
          <physvol name="Si_a_ring3_physvol1" copynumber="(a-1)*14+17">
          <physvol name="Si_a_ring3_physvol2" copynumber="(a-1)*14+24">
          <physvol name="Si_a_ring4_physvol1" copynumber="(a-1)*14+18">
          <physvol name="Si_a_ring4_physvol2" copynumber="(a-1)*14+25">
          <physvol name="Si_a_ring5_physvol1" copynumber="(a-1)*14+19">
          <physvol name="Si_a_ring5_physvol2" copynumber="(a-1)*14+26">
          <physvol name="Si_a_ring6_physvol1" copynumber="(a-1)*14+20">
          <physvol name="Si_a_ring6_physvol2" copynumber="(a-1)*14+27">
          <physvol name="Si_b_center_physvol1" copynumber="(b-1)*14+2*ecal_A_layers*7+14">
          <physvol name="Si_b_center_physvol2" copynumber="(b-1)*14+2*ecal_A_layers*7+21">
          <physvol name="Si_b_ring1_physvol1" copynumber="(b-1)*14+2*ecal_A_layers*7+15">
          <physvol name="Si_b_ring1_physvol2" copynumber="(b-1)*14+2*ecal_A_layers*7+22">
          <physvol name="Si_b_ring2_physvol1" copynumber="(b-1)*14+2*ecal_A_layers*7+16">
          <physvol name="Si_b_ring2_physvol2" copynumber="(b-1)*14+2*ecal_A_layers*7+23">
          <physvol name="Si_b_ring3_physvol1" copynumber="(b-1)*14+2*ecal_A_layers*7+17">
          <physvol name="Si_b_ring3_physvol2" copynumber="(b-1)*14+2*ecal_A_layers*7+24">
          <physvol name="Si_b_ring4_physvol1" copynumber="(b-1)*14+2*ecal_A_layers*7+18">
          <physvol name="Si_b_ring4_physvol2" copynumber="(b-1)*14+2*ecal_A_layers*7+25">
          <physvol name="Si_b_ring5_physvol1" copynumber="(b-1)*14+2*ecal_A_layers*7+21">
          <physvol name="Si_b_ring5_physvol2" copynumber="(b-1)*14+2*ecal_A_layers*7+26">
          <physvol name="Si_b_ring6_physvol1" copynumber="(b-1)*14+2*ecal_A_layers*7+20">
          <physvol name="Si_b_ring6_physvol2" copynumber="(b-1)*14+2*ecal_A_layers*7+27">
          <physvol name="Si_c_center_physvol1" copynumber="(c-1)*14+2*ecal_A_layers*7+2*ecal_B_layers*7+14">
          <physvol name="Si_c_center_physvol2" copynumber="(c-1)*14+2*ecal_A_layers*7+2*ecal_B_layers*7+21">
          <physvol name="Si_c_ring1_physvol1" copynumber="(c-1)*14+2*ecal_A_layers*7+2*ecal_B_layers*7+15">
          <physvol name="Si_c_ring1_physvol2" copynumber="(c-1)*14+2*ecal_A_layers*7+2*ecal_B_layers*7+22">
          <physvol name="Si_c_ring2_physvol1" copynumber="(c-1)*14+2*ecal_A_layers*7+2*ecal_B_layers*7+16">
          <physvol name="Si_c_ring2_physvol2" copynumber="(c-1)*14+2*ecal_A_layers*7+2*ecal_B_layers*7+23">
          <physvol name="Si_c_ring3_physvol1" copynumber="(c-1)*14+2*ecal_A_layers*7+2*ecal_B_layers*7+17">
          <physvol name="Si_c_ring3_physvol2" copynumber="(c-1)*14+2*ecal_A_layers*7+2*ecal_B_layers*7+24">
          <physvol name="Si_c_ring4_physvol1" copynumber="(c-1)*14+2*ecal_A_layers*7+2*ecal_B_layers*7+18">
          <physvol name="Si_c_ring4_physvol2" copynumber="(c-1)*14+2*ecal_A_layers*7+2*ecal_B_layers*7+25">
          <physvol name="Si_c_ring5_physvol1" copynumber="(c-1)*14+2*ecal_A_layers*7+2*ecal_B_layers*7+19">
          <physvol name="Si_c_ring5_physvol2" copynumber="(c-1)*14+2*ecal_A_layers*7+2*ecal_B_layers*7+26">
          <physvol name="Si_c_ring6_physvol1" copynumber="(c-1)*14+2*ecal_A_layers*7+2*ecal_B_layers*7+20">
          <physvol name="Si_c_ring6_physvol2" copynumber="(c-1)*14+2*ecal_A_layers*7+2*ecal_B_layers*7+27">
          <physvol name="Si_d_center_physvol1" copynumber="(d-1)*14+2*ecal_A_layers*7+2*ecal_B_layers*7+2*ecal_C_layers*7+14">
          <physvol name="Si_d_center_physvol2" copynumber="(d-1)*14+2*ecal_A_layers*7+2*ecal_B_layers*7+2*ecal_C_layers*7+21">
          <physvol name="Si_d_ring1_physvol1" copynumber="(d-1)*14+2*ecal_A_layers*7+2*ecal_B_layers*7+2*ecal_C_layers*7+15">
          <physvol name="Si_d_ring1_physvol2" copynumber="(d-1)*14+2*ecal_A_layers*7+2*ecal_B_layers*7+2*ecal_C_layers*7+22">
          <physvol name="Si_d_ring2_physvol1" copynumber="(d-1)*14+2*ecal_A_layers*7+2*ecal_B_layers*7+2*ecal_C_layers*7+16">
          <physvol name="Si_d_ring2_physvol2" copynumber="(d-1)*14+2*ecal_A_layers*7+2*ecal_B_layers*7+2*ecal_C_layers*7+23">
          <physvol name="Si_d_ring3_physvol1" copynumber="(d-1)*14+2*ecal_A_layers*7+2*ecal_B_layers*7+2*ecal_C_layers*7+17">
          <physvol name="Si_d_ring3_physvol2" copynumber="(d-1)*14+2*ecal_A_layers*7+2*ecal_B_layers*7+2*ecal_C_layers*7+24">
          <physvol name="Si_d_ring4_physvol1" copynumber="(d-1)*14+2*ecal_A_layers*7+2*ecal_B_layers*7+2*ecal_C_layers*7+18">
          <physvol name="Si_d_ring4_physvol2" copynumber="(d-1)*14+2*ecal_A_layers*7+2*ecal_B_layers*7+2*ecal_C_layers*7+25">
          <physvol name="Si_d_ring5_physvol1" copynumber="(d-1)*14+2*ecal_A_layers*7+2*ecal_B_layers*7+2*ecal_C_layers*7+19">
          <physvol name="Si_d_ring5_physvol2" copynumber="(d-1)*14+2*ecal_A_layers*7+2*ecal_B_layers*7+2*ecal_C_layers*7+26">
          <physvol name="Si_d_ring6_physvol1" copynumber="(d-1)*14+2*ecal_A_layers*7+2*ecal_B_layers*7+2*ecal_C_layers*7+20">
          <physvol name="Si_d_ring6_physvol2" copynumber="(d-1)*14+2*ecal_A_layers*7+2*ecal_B_layers*7+2*ecal_C_layers*7+27">
```

awhitbeck, 2017-08-08:
@ASPatterson Indeed this is a bug.  You fix in the latest commit of iss340 looks correct to me.

omar-moreno, 2017-08-09:
This issue was resolved by commit https://github.com/LDMXAnalysis/ldmx-sw/pull/338/files.

closed 2017-08-09 by omar-moreno