LDMX-Software/ldmx-sw #424: drop rules remove collections before event processors are executed

awhitbeck, 2018-01-20:
I think this is an unintended feature and instead, we want the ability to process information from collections and then discard them in the output file.  

I ran into this while I was debugging an empty collection issue related to #412, where I sort sim hits, create a new sim hit collection and then drop the old sim hits.  I think its a pretty easy fix:

move [these](https://github.com/LDMXAnalysis/ldmx-sw/blob/1bf50604b3c203eaf274e957568fa4178dd0ad5e/Framework/src/Process.cxx#L82-L84) lines to just before [this](https://github.com/LDMXAnalysis/ldmx-sw/blob/master/Framework/src/Process.cxx#L143) line.

This at least solves my problem with #412, but I am not sure if there are other undesirable consequences. I was hoping @jmmans could comment.

jmmans, 2018-01-22:
Does this actually cause the data to be dropped?

awhitbeck, 2018-01-22:
Yeah.  Did you have some other expectation?

closed 2019-05-05 by omar-moreno