LDMX-Software/ldmx-sw #482: LHE Reader should use an XML parser instead of reading the document line by line

omar-moreno, 2018-08-01:
The way it's written now, if there is an extra node in the tree, it causes our parser to crash.  With an XML parser, we would be able to directly access the event node so extra information wouldn't matter.

tomeichlersmith, 2019-07-16:
I found a [LHE Reader](http://home.thep.lu.se/~leif/LHEF/LHEF_8h_source.html) that is definitely more robust that our current implementation. The difficulty is that I can't figure out how to get the primary vertex right now. If we use this reader, we might need to calculate the primary vertex from distance/momentum information.

Sidenote: This reader doesn't use an XML parser either, but it seems specialized and the example LHE files to use for input match (in form) to the LHE files I've used.

omar-moreno, 2019-07-16:
The primary vertex is something that is added to each event. So you
basically want to set up the reader in such a way that if it finds this
tag, it processes it and sets the vertex. Otherwise, the file can be parsed
as usual.

Where did you find the new LHE reader?

tomeichlersmith, 2019-07-16:
Of course, immediately after posting this, I found that the LHE Reader above doesn't parse for vertex information. I'm going to see if deriving a class and adding our own parser searching for the `#vertex` tag would be able to do what we want.

I found this LHE Reader after searching around for an xml schema for LHE files. It's included in the [HepMC Library](http://hepmc.web.cern.ch/hepmc/).

omar-moreno, 2019-07-16:
Oh nice. Great, thanks Tom.

tomeichlersmith, 2019-07-16:
So this LHE Reader works pretty well, but there are a few issues that need to be addressed:

1. The Reader throws an error if it is unable to read a header block containing run information.
This problem is easy to "solve" since this reader is confined to a header file. I commented out the code that attempts to read the header block, so we manually skip that. This might not be a good solution, maybe change the `runtime_error` to a simple warning?

2. The Reader throws an error if the lines defining the particles are not immediately following the lines defining the event.
This problem is also easy to address if we remember to do this when generating LHE files in the future. The LHE files I have access to have the `#vertex` flag in between the event line and the particle lines, but moving it below the particle lines fixes the problem. I'm not sure how wide spread this problem is. At UMN, we shared a python script that generated LHE files for us, and I can edit that to prevent this problem in the future. Is it easy for everyone else to move the `#vertex` flag to the last line of the event definition?

3. The Reader doesn't know the `#vertex` flag.
This is actually easier to solve than I thought! The Reader puts all of the information it doesn't recognize that is at the end of the event definition into a member string called `junk`. I can simply parse this member string for the vertex information.

tomeichlersmith, 2019-07-16:
I have been able to get the same functionality that we had before. The run still terminates if more events are requested than are listed in the LHE file, but the events that are listed in the LHE file are simulated regularly. I addressed the issues above as follows:
1. I commented out the initialization function in the constructor that we use. This means it doesn't attempt to read `<init>...</init>` or even `<header>...</header>` blocks, which only works well for LHE files that only contain a list of `<event>` blocks. 
2. There isn't really a way to fix this for the LHE Reader I'm using. This "fix" would force a new style for our LHE files where the `#vertex` flag is in the line above the closing `</event>` tag. We could dig into the `LHEF.h` and add the `#vertex` combing there, but I did not do that in this first pass.
3. I input the leftovers string into a stringstream that I can then parse looking for the vertex flag.

tomeichlersmith, 2019-07-16:
Found some [better documenation](https://gitlab.cern.ch/hepmc/HepMC3/blob/master/doc/doxygen/LHEF.txt) on `LHEF.h` that I just want to save here in case we decide to standardize how we use LHE files.

omar-moreno, 2019-09-13:
Will close this for now but should definitely consider standardizing in the near future.

closed 2019-09-13 by omar-moreno