LDMX-Software/ldmx-sw #529: Use TTreeReader when processing files

omar-moreno, 2019-04-09:


tomeichlersmith, 2019-07-08:
I've been looking into implementing this, and I don't think there is an easy way to do this without a redesign of the reading framework.

In the output `EventFile`, the output tree is initially cloned from the input tree (copying the branch addresses) so that if the input tree branch has a status allowing it to be read, it will be added to the output tree (and thus the output file). `TTreeReader` doesn't have an interface for setting/copying branch addresses, so we would need to leave this as is.

We could use a `TTreeReader` to obtain the objects from the input tree, but then the member variable `EventImpl::objects_` would need to be separated into `TTreeReaderValue`s for the input reader and general objects that are being created by the process. This might be helpful because it would hard code in the separation between data that is read in and data that is created during processing, but we would have to check both of these new object maps whenever we previous only checked the one.

All this being said, I'm wondering what you were interested in with the tree reader. The documentation implies it has improved efficiency, so we could put in the work for a redesign that could increase the reading efficiency. I could also have a misunderstanding about how either the framework or the reader functions, so this could be an unhelpful comment.

omar-moreno, 2019-07-08:
This issue was more of a remainder to myself that we would want to make use
of TTreeReader at some point.  As you said, it not only improves the
efficiency in reading files, but also simplifies concurrent access.   Of
course, making use of TTreeReader will require a rewrite of a large portion
of the framework.

I say, for now, this is something we should keep in mind but not implement
until we can come up with a clear approach.  I'm happy to talk more about
this.

closed 2020-06-22 by omar-moreno