LDMX-Software/ldmx-sw #568: Patch memory leaks across processors

tomeichlersmith, 2019-07-05:
Most processors have some form of a memory leak causing them to use up more RAM than needed during processing. Simply need to go through the processors and put in nicely placed `delete` commands to make sure everyone is cleaning up after themselves.

omar-moreno, 2019-07-05:
I would make use of smart pointers when possible.

tomeichlersmith, 2019-07-08:
On the branch for this issue, I have been able to update most of the pointers in `EventProc` and some of the pointers in `Ecal`, `Event`, and `Framework` to smart pointers. The difficulty I'm running in to now is how pointers interact with ROOT. I don't want to use the `get` function to obtain the bare pointer from a smart pointer because that leads to undefined behaviour especially in our case where the smart pointer might go out of scope before ROOT is done with the object, so I'm leaving those pointers as bare.

Any suggestions? Or are we stuck with this solution because of how ROOT operates?

omar-moreno, 2019-07-08:
Unfortunately, I think we are stuck with this solution to accommodate
ROOT.  As long as we delete those, I say it's fine for now.

tomeichlersmith, 2019-07-08:
Sounds good, I will create a pull request for review.

omar-moreno, 2019-09-13:
Solved by PR #569.

closed 2019-09-13 by omar-moreno