LDMX-Software/ldmx-sw #574: gdml to TGeo friendly gdml

tomeichlersmith, 2019-07-15:
Currently, the ROOT implementation of gdml parsing struggles with the more complicated gdml parts (like loops and replica volumes). We can get around these struggles by having a utility translate the original gdml files into a ROOT-friendly gdml file.

I hope to use Geant itself to do this translation by importing the detector geometry into Geant and then exporting it back into one file (maybe with some specific write options).

tomeichlersmith, 2019-07-15:
I have been able to write an executable in the `DetDescr` module, but ROOT is still unable to handle the magnet. ROOT doesn't know the `triangular` tag for some reason, but going through Geant4 does produce less errors when importing to ROOT than trying to input the gdml files directly (and drawing after importing can be done).

The error `TGeoManager::Import` throws is
```
Error: Unsupported GDML Tag Used :triangular. Please Check Geometry/Schema.
^^^ Copy of this error for each triangular in magnet.gdml ^^^
Solid: coil_10x1f792c0, Not Yet Defined!
Solid: coil_20x36fe060, Not Yet Defined!
```
It seems like ROOT can't define the coils because it doesn't know how to draw the triangular tessellation. I tried using the schema defined at the header of the `detector.gdml` file in the `G4GDMLParser::Write` command, but that didn't change the output. Looking in to if this is a ROOT incompatibility or if I can include a specific schema to tell ROOT how `triangular` works.

omar-moreno, 2019-07-15:
Yeah, this is the same issue I was running into before. That's why I
created the geometry without a magnet in it. Try using that one.

tomeichlersmith, 2019-07-15:
It works well without the magnet. After using the "translation" executable, I can draw the geometry in ROOT simply with
```
gSystem->Load("libGeom"); gSystem->Load("libGdml");
TGeoManager::Import("translated-file.gdml");
gGeoManager->Draw("ogl");
```
I feel like `triangular` is simple enough that ROOT would be able to handle it, so I'm going to try to figure out if there is a schema I can use that would help.

tomeichlersmith, 2019-07-15:
As of [April 2017](https://root-forum.cern.ch/t/errors-importing-gdml-tessellated-solid-with-root/24313/2), tesselated solids were not supported in ROOT (version <= 6.09/02). I have not been able to find any patches in the release notes for each version since then, so I think we are stuck.

omar-moreno, 2020-06-22:
Porting things to dd4hep will eliminate the need for this.

closed 2020-06-22 by omar-moreno