LDMX-Software/ldmx-sw #575: HcalDigiProducer calculates position of hits incorrectlye for side hcal

tomeichlersmith, 2019-07-15:
`HcalDigiProducer` declares the floats `curr_xpos` and `curr_ypos` in order to be manipulated later by the smearing procedure done for the hits in the *back* hcal. This means that for the *side* hcal, the x and y positions of the hit are set to whatever value is in memory for those variables.

This is a simple fix: define the current x and y positions as the x and y position of the sim hit so the position is correct for the side hcal.

This also brings up a question of if we want to do any kind of smearing procedure for the side hcal (similar to what we do for the back hcal).

bechenard, 2019-07-15:
HI Tom,

Yes, I caught this and I'm working on this right now! I've cleaned up 
the code and fixed several other issues. I'll get back to you soon and 
we can coordinate the changes.

Cheers,

Bertrand

tomeichlersmith, 2019-07-15:
I made a branch `iss575` working on HcalDigiProducer. Feel free to look at what I've done and add your changes to it.

tomeichlersmith, 2019-07-30:
Bertrand has a more precise/up-to-date branch for Issue #576 . So I'm closing this issue.

closed 2019-07-30 by tomeichlersmith