LDMX-Software/ldmx-sw #582: The layer weights in EcalDigiProducer are incorrect for several geometries (v9).

tomeichlersmith, 2019-07-30:
I have uploaded a script to ldmx-sw-scripts that can calculate the layer weights given a list of the material layers: [calculate_layer_weights.py](https://github.com/LDMX-Software/ldmx-sw-scripts/blob/master/utils/calculate_layer_weights.py). I used this to calculate the layer weights for the v9 geometry and the v12 geometry (on branch iss577). Both of these weights were significantly different than the layer weights in `EcalDigiProducer` right now. Allowing both the layer weights and the second order correction to be given as config parameters allows for changing geometries (and doing geometric studies) to be much easier.

omar-moreno, 2019-07-30:
This is probably something we want in before we re-run the reconstruction
to update the BDT.  Is there a plan to get this in soon?

tomeichlersmith, 2019-07-30:
I have created a PR #583 that offers a fix for this issue. If you want to double check the layer weights calculation, you can look at the script I used in ldmx-sw-scripts: [calculat_layer_weights.py](https://github.com/LDMX-Software/ldmx-sw-scripts/blob/master/utils/calculate_layer_weights.py).

omar-moreno, 2019-07-30:
Great.  We probably want to do some physics validation to really understand what impact this has.  If it's significant, then we will likely need to retrain the BDT. @vdutta should be looped in here.