LDMX-Software/ldmx-sw #584: Detector Geometry Isolation

tomeichlersmith, 2019-07-31:
Currently, a lot of different parts of the geometry description are hard-coded in various producers/analyzers/classes. Isolating the geometry description to one class in the DetDescr module would allow for all modules to access the same geometry details, making it standard across the framework. Additionally, keeping the geometry in one place will make it easier to update (and maybe automatically load from a gdml description) in the future.

I started the class on this branch to help with the event display development, but I think it deserves to be separate because it could be helpful for everyone.

Right now, it has the HCAL and ECAL geometries with some accessor functions that are specialized for the event display. At the end of the implementation file, a namespace wide instance is created called `DETECTOR_GEOMETRY` which can be used any where the header file is included.

bechenard, 2019-07-31:
Hi Tom,

Great, that is going to be useful.

Bertrand

omar-moreno, 2019-08-22:
Before doing anything here, I would like to see a presentation with an outline on approach and the class hierarchy you are envisioning.  A lot of work has already been done on this and I want to make sure that  whatever classes you are going to push wont clash with what has already been done. 

Also, using a singleton is something I want to avoid for now.