LDMX-Software/ldmx-sw #814: Add parent ID to Calorimeter hit contributions.

omar-moreno, 2020-06-23:
**Is your feature request related to a problem? Please describe.**
Currently, there is no easy way to associate the MC particle contribution to an ECal sim hit to its parent particle.  This makes it difficult to associate showers to different incident particles. 

**Describe the solution you'd like**
Add parentID to the contributions.

tomeichlersmith, 2020-06-23:
Since the trackID is already apart of the SimCalorimterHit contribs, what about storage of the full ancestry information? I'm worried that the parent ID won't be much help, especially deeper in the shower where the parent would still be removed from more important particles and not be stored.

omar-moreno, 2020-06-23:
I'm afraid that doing that would blow up an event and, in general, isn't needed. Just making sure all hit contributions are assigned a parent particle will go a long way in improving what's already there while also minimally increasing the event size.

vdutta, 2020-06-23:
Just to be clear, @omar-moreno, would the parent ID be for the original ancestor (e.g. the recoil electron), or for the immediate parent of the particle that might have been produced in a shower?

omar-moreno, 2020-06-23:
original.