LDMX-Software/ldmx-sw #816: Configure logging from the python configuration

tomeichlersmith, 2020-06-24:
This will simply add some parameters to the `Process` python class that will be passed to the boost logging.

We put in the necessary framework for logging with PR #811 , and this is the next step for it to be usable.

My plan is to clean up the `fire.cxx` file so that the configuration stuff has its own `try...catch` branch that is outside of the logger, and then the rest of the running will be within the `Process` C++ class which puts all of its messages into the logger.

tomeichlersmith, 2020-07-02:
Alright, I've gotten this to a good spot. @omar-moreno My question is: do we want to have the event bus printing go into the logging? 

Pros: Everything goes into logger no matter what.
Cons: Would need to change all of the `Print` methods to work with an input `ostream`.

I don't think anyone uses the `event.Print` function unless they are trying to debug something, so I don't think it is important to have in the logger (and I don't want to have to put in the time).

closed 2020-07-02 by tomeichlersmith

omar-moreno, 2020-07-02:
@tomeichlersmith Just saw the comment.  I would put it into the logger because, as you say, the common use case is likely when debugging. You can do this in a separate issue if you want.