LDMX-Software/ldmx-sw #826: barID in trigScintDigi set to 1023

awhitbeck, 2020-06-26:
For the upstream and downstream arrays there seems to be a problem with barID:

[image]

These numbers should never be larger than 50. I am including the configuration file that I used to generate these vents.  I see this in a few dozen events out of 10k. 

```
from LDMX.Framework import ldmxcfg
from LDMX.Detectors.makePath import makeDetectorPath
from LDMX.SimApplication import generators
from LDMX.SimApplication import examples

p=ldmxcfg.Process("sim")

p.libraries.append("libSimApplication.so")
p.libraries.append("libBiasing.so")
p.libraries.append("libEventProc.so")

mySim = ldmxcfg.Producer( 'mySim' , 'ldmx::Simulator' )
mySim.parameters[ "detector" ] = makeDetectorPath( "ldmx-det-v12" )

mpgGen = generators.multi( "mgpGen" ) # this is the line that actually creates the generator
mpgGen.parameters[ "vertex" ] = [ -27.926, 0., -700 ] # mm
mpgGen.parameters[ "nParticles" ] = 1
mpgGen.parameters[ "pdgID" ] = 11
mpgGen.parameters[ "enablePoisson" ] = True
mpgGen.parameters[ "momentum" ] = [ 380., 0., 3981.909 ] #MeV

mySim.parameters['generators'] = [ mpgGen ]

mySim.parameters["beamSpotSmear"] = [20., 80., 0.]
mySim.parameters[ "runNumber" ] = 9001
mySim.parameters[ "verboseLevel" ] = 10

p.sequence=[mySim]

from LDMX.EventProc.trigScintDigis import *
trigScintDigis.parameters["pe_per_mip"]=100.
trigScintDigisDn.parameters["pe_per_mip"]=100.
trigScintDigisTag.parameters["pe_per_mip"]=100.

p.sequence.extend([trigScintDigis,
                   trigScintDigisDn,
                   trigScintDigisTag])

from LDMX.EventProc.trackerHitKiller import trackerHitKiller
findableTrack = ldmxcfg.Producer("findable", "ldmx::FindableTrackProcessor")
p.sequence.extend([trackerHitKiller,findableTrack])

p.outputFiles=['test.root']

p.maxEvents = 10000
p.logFrequency = 1000
```

tomeichlersmith, 2020-06-26:
1023 = 2^10 - 1, so I'm guessing there is an overflow somewhere inside of `TrigScintID` when doing some bit-shifting nonsense

bryngemark, 2020-06-26:
yeah this to me looks like initializing to -1 and then never setting it properly

omar-moreno, 2020-06-26:
Did anything change since we tested this last? I checked the ID's for
upstream and downstream before and they were correct. It's odd that it's
showing up now.

bryngemark, 2020-06-26:
if it's true that this doesn't happen in the tagger module but in upstream and downstream modules, i'd check specifically that the module ID gets set properly.

awhitbeck, 2020-06-26:
I think this only happens when no particle passes through the array, but that is only based on a few anecdotal observations. There shouldn't be any cases where no particle passes through the tagger array.

omar-moreno, 2020-06-26:
> I think this only happens when no particle passes through the array, but
> that is only based on a few anecdotal observations. There shouldn't be any
> cases where no particle passes through the tagger array.

OK, if that's the case, this makes more sense.

bryngemark, 2020-06-26:
Right... so these are all noise hits in that case?
The barID setting for noise hits assumes that the module number has been set. I see that this happens inside the real simhit loop. So that should probably be changed to accommodate the case when a module only has noise hits.

jmmans, 2020-11-05:
Can this issue be closed?

bryngemark, 2020-11-05:
No fix for this has been pushed yet. I did think about it yesterday but I think only ugly (collection name parsing) solutions can be implemented until we move to processing all three modules in one single collection. 

So, I'd say, can't be closed.

tvami, 2025-08-20:
I think this issue is resolved, at least when I print out cases for noise, I dont see them at 1023
```
[ TrigScintDigiVerifier ] 4  info: There are 1 ts digis in this event
[ TrigScintDigiVerifier ] 4  info: Digi with raw ID 134218752 and bar ID 0 has energy 6.9425
[ TrigScintDigiVerifier ] 5  info: There are 1 ts digis in this event
[ TrigScintDigiVerifier ] 5  info: Digi with raw ID 134218752 and bar ID 0 has energy 78.8158
[ TrigScintDigiVerifier ] 6  info: There are 2 ts digis in this event
[ TrigScintDigiVerifier ] 6  info: Digi with raw ID 134218752 and bar ID 0 has energy 89.5097
[ TrigScintDigiVerifier ] 6  info: Digi with raw ID 134218793 and bar ID 41 is flagged as noise, skipping
[ TrigScintDigiVerifier ] 7  info: There are 1 ts digis in this event
[ TrigScintDigiVerifier ] 7  info: Digi with raw ID 134218752 and bar ID 0 has energy 1.57889
[ TrigScintDigiVerifier ] 8  info: There are 1 ts digis in this event
[ TrigScintDigiVerifier ] 8  info: Digi with raw ID 134218752 and bar ID 0 has energy 1.04642
[ TrigScintDigiVerifier ] 9  info: There are 2 ts digis in this event
[ TrigScintDigiVerifier ] 9  info: Digi with raw ID 134218752 and bar ID 0 has energy 7.0558
[ TrigScintDigiVerifier ] 9  info: Digi with raw ID 134218777 and bar ID 25 is flagged as noise, skipping
[ TrigScintDigiVerifier ] 10  info: There are 2 ts digis in this event
[ TrigScintDigiVerifier ] 10  info: Digi with raw ID 134218752 and bar ID 0 has energy 1.32512
[ TrigScintDigiVerifier ] 10  info: Digi with raw ID 134218790 and bar ID 38 is flagged as noise, skipping
[ TrigScintDigiVerifier ] 11  info: There are 4 ts digis in this event
[ TrigScintDigiVerifier ] 11  info: Digi with raw ID 134218752 and bar ID 0 has energy 4.51475
[ TrigScintDigiVerifier ] 11  info: Digi with raw ID 134218765 and bar ID 13 is flagged as noise, skipping
[ TrigScintDigiVerifier ] 11  info: Digi with raw ID 134218786 and bar ID 34 is flagged as noise, skipping
[ TrigScintDigiVerifier ] 11  info: Digi with raw ID 134218793 and bar ID 41 is flagged as noise, skipping
[ TrigScintDigiVerifier ] 12  info: There are 2 ts digis in this event
[ TrigScintDigiVerifier ] 12  info: Digi with raw ID 134218752 and bar ID 0 has energy 25.19
[ TrigScintDigiVerifier ] 12  info: Digi with raw ID 134218797 and bar ID 45 is flagged as noise, skipping
[ TrigScintDigiVerifier ] 13  info: There are 1 ts digis in this event
[ TrigScintDigiVerifier ] 13  info: Digi with raw ID 134218752 and bar ID 0 has energy 3.66013
[ TrigScintDigiVerifier ] 14  info: There are 2 ts digis in this event
[ TrigScintDigiVerifier ] 14  info: Digi with raw ID 134218752 and bar ID 0 has energy 7.63686
[ TrigScintDigiVerifier ] 14  info: Digi with raw ID 134218776 and bar ID 24 is flagged as noise, skipping
[ TrigScintDigiVerifier ] 15  info: There are 3 ts digis in this event
[ TrigScintDigiVerifier ] 15  info: Digi with raw ID 134218752 and bar ID 0 has energy 12.1276
[ TrigScintDigiVerifier ] 15  info: Digi with raw ID 134218759 and bar ID 7 is flagged as noise, skipping
[ TrigScintDigiVerifier ] 15  info: Digi with raw ID 134218780 and bar ID 28 is flagged as noise, skipping
[ TrigScintDigiVerifier ] 16  info: There are 2 ts digis in this event
[ TrigScintDigiVerifier ] 16  info: Digi with raw ID 134218752 and bar ID 0 has energy 11.4662
[ TrigScintDigiVerifier ] 16  info: Digi with raw ID 134218777 and bar ID 25 is flagged as noise, skipping
[ TrigScintDigiVerifier ] 17  info: There are 2 ts digis in this event
[ TrigScintDigiVerifier ] 17  info: Digi with raw ID 134218752 and bar ID 0 has energy 5.71929
[ TrigScintDigiVerifier ] 17  info: Digi with raw ID 134218781 and bar ID 29 is flagged as noise, skipping
[ TrigScintDigiVerifier ] 18  info: There are 3 ts digis in this event
[ TrigScintDigiVerifier ] 18  info: Digi with raw ID 134218752 and bar ID 0 has energy 12.8837
[ TrigScintDigiVerifier ] 18  info: Digi with raw ID 134218769 and bar ID 17 is flagged as noise, skipping
[ TrigScintDigiVerifier ] 18  info: Digi with raw ID 134218791 and bar ID 39 is flagged as noise, skipping
```

It is interesting to see that the non-noise hit is always on bar ID 0 ?

closed 2025-08-20 by tvami