LDMX-Software/ldmx-sw #851: Add python module to load Event dictionary

tomeichlersmith, 2020-07-08:
A lot of users use python to load event files and do some quick analyses and/or just investigate the event file. Adding a python module in the `Event` module that can do the complex nonsense of loading the Event dictionary into ROOT will be helpful.

tomeichlersmith, 2020-07-08:
From @jmlazaro25 :

jmlazaro25, 2020-07-08:
Testing bdtTreeMaker.py with the pytools derivitive

self.evHeader = ROOT.ldmx.EventHeader()
  File "/deps/cernroot/lib/ROOT.py", line 674, in __getattr2
    raise AttributeError( name )
AttributeError: ldmx

tomeichlersmith, 2020-07-08:
It looks like it has to do with PyROOT being built with python2 instead of python3:
```
tom@zuko:~/ldmx/ldmx-sw$ ldmx python3
Python 3.6.9 (default, Apr 18 2020, 01:56:04) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from LDMX.Event import dictionary
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/tom/ldmx/ldmx-sw/install/lib/python/LDMX/Event/dictionary.py", line 12, in <module>
    import ROOT
  File "/deps/cernroot/lib/ROOT.py", line 24, in <module>
    import cppyy
  File "/deps/cernroot/lib/cppyy.py", line 61, in <module>
    import libPyROOT as _backend
ImportError: dynamic module does not define module export function (PyInit_libPyROOT)
>>> quit()
```

PR #852 (2020-07-10): import dictionary file inside Event module
This resolves #851 

Marked as draft because I need to test it and make sure it works.
closed 2020-07-10 by omar-moreno