LDMX-Software/ldmx-sw #869: Compile EventDisplay outside container

tomeichlersmith, 2020-07-21:
This has been requested by several people and will be very helpful. It will require to be built outside of the container, but it won't need Geant4 or XercesC. The only dependency within ldmx-sw are Event and DetDescr, so (hopefully) pulling out the EventDisplay module could be done rather easily.

I am planning on pulling out the EventDisplay module into a separate repository that is then pulled into ldmx-sw. I think this makes sense because it needs to be outside the container, and maybe we can pull it into the container later.

omar-moreno, 2020-07-21:
I think you should pull it out and leave it as it's own repo without pulling it into ldmx-sw.  It will be dead anyways once dd4hep gets put it.

tomeichlersmith, 2020-07-21:
The issue is that the EventDisplay requires (at least) the Event, DetDescr, and Framework modules so that it can correctly readin the Event Bus. If we also pull out the Event and DetDescr modules, then we could have the EventDisplay be a completely separate repo, but if we don't do that, then it needs to be pulled into ldmx-sw.

omar-moreno, 2020-07-21:
Then I don't know why you would pull it out into it's own repo? What advantage does pulling this out of ldmx-sw have?

omar-moreno, 2020-07-21:
With the changes to CMakeLists that I'm going to push tonight, there wont be a need to install Geant4 or Xerces if all users want to do is build the EventDisplay module.

tomeichlersmith, 2020-07-21:
Ok, then I won't need to pull it out of ldmx-sw. I will just need to tweak it so people can do an out-of-container build (if they have ROOT with the correct dependencies installed).

omar-moreno, 2020-07-21:
I guess I'm not understanding the issue.  Is it that you want people to make use of dependencies installed within the container to build the EventDisplay out of the container?

omar-moreno, 2020-07-21:
Why not just have an EventDisplay container?

tomeichlersmith, 2020-07-21:
So the issue is that the EventDisplay cannot be built inside of the container. Setting up a container that can have access to the host's screen is complicated and something I want to avoid right now. Users will not be able to use the in-container dependencies to compile it; they will still need an install of ROOT that has the Eve modules built.

Your cmake changes actually get us most of the way towards being able to compile EventDisplay outside of the container without having to compile the other modules, so this issue would be mainly focused on (1) making sure that can be done, and (2) updating the EventDisplay for v12 geometry.

omar-moreno, 2020-07-21:
OK, makes more sense now.  It's probably safe to assume that most users will have a copy of ROOT installed locally, so once you can build the module independently of others, then pulling it wont make a difference. 

That said, if you feel like it makes maintaining easier, then go ahead and pull it out.

tomeichlersmith, 2020-07-22:
I like your idea of having a event display specific container, so I am going to look into doing that.

Some notes
- [Execute a Firefox Browser in a Docker Container](http://vmwareinsight.com/Articles/2020/5/5803001/Execute-a-Firefox-Browser-in-a-Docker-Container-with-Graphical-Interface)
- [GUI-Enabled Docker Container](https://www.cbtechinc.com/desktop-docker-1-linux-graphical-containers/)

Necessities
- [ ] ROOT with Eve modules built
      - Doesn't happen automatically, probably need to install more dependencies
- [ ] EventDisplay module built and installed
- [ ] Connection to host machines X11 authority

closed 2020-08-24 by omar-moreno