LDMX-Software/ldmx-sw #873: Improve Realism of Cellular Layout in EcalHexReadout

tomeichlersmith, 2020-07-25:
We are not very familiar with how the hex modules will be built, but we know several details that can be implemented in EcalHexReadout to improve the realism of the cellular layout that aren't implemented right now.

Currently, the cellular layout is a uniform hexagonal grid across the module: [Cell_ID_Cell_Position_Map.pdf](https://github.com/LDMX-Software/ldmx-sw/files/4976162/Cell_ID_Cell_Position_Map.pdf)

This positioning has several flaws that need to be rectified:
- [ ] Only 397 cells are numbered, there should be 432
- [ ] Mousebites: The cells do not penetrate into the corners of the module hexagon to leave room for mounting
- [ ] All the cells are shifted by one half of a cell width so that the center of the module is positioned on the border between cells instead of at the center of a cell (198)
- [ ] There needs to be a cell ID --> trigger cell ID mapping. The trigger cells are groups of 9 (3x3) DAQ cells that are added together during the trigger digitization. The outline for them can be seen [in these slides](http://homepages.spa.umn.edu/~eichl008/digi-pipeline/bkgrd/FD_HGCROC3_190702_TowardsV3.pdf).
- [ ] Incorporate a cell ID <-> UV coordinate translation. This will be helpful for defining trigger cell groupings as well as defining neighboring cells. (UV coordinates are coordinate axes parallel to two adjacent sides of the hexagon, so axes that make an angle of 120 degrees)
- [ ] The cell ID with respect to the x-coordinate will flip between sensitive layers because the modules will be flipped when mounted to make the full layer "sandwich".

@omar-moreno This is _not_ something I can implement in the short-term, so don't expect this change before the next production. I just want to save my notes so that I don't lose them.

@jmmans Can you attach some documents about the HGCAL module geometry? I remember you showing me some diagrams that helped show the cellular layout, but I can't find them.

jmmans, 2020-07-26:
I've attached the high-density sensor information which is available.

[image]

Tom -- you have a script which makes the map drawing.  Does this script work "in general"?  I'm thinking to try to address the 432 cell issue and the shifted cell issue, which are tied together.

tomeichlersmith, 2020-07-27:
Not completely general. Right now it prints the `TH2Poly` histogram that is used to define the cells and then draws the module hexagon centered at the origin on top of that histogram, so the printer would work if you are doing the increasing number and shifting within the `ecalMap_` member variable. I could see a solution where the shifting occurs "before" using the TH2Poly to get a cell (e.g. by subtracting a constanct from the y-coordinate), and in that case, the printing would not take the shift into effect.

You can see what I do to print it by looking at [DetDescr/src/print_ecal_hex_readout.cxx on the ecal-digi branch](https://github.com/LDMX-Software/ldmx-sw/blob/ecal-digi/DetDescr/src/print_ecal_hex_readout.cxx); feel free to modify this executable if you need to to print the _actual_ cell lay out and not the TH2Poly.

tomeichlersmith, 2021-06-18:
This was an overly-ambitious issue and has already accomplished most of its goals (a long time ago). I have moved the last two goals to #923 so I don't forget them as we move forward.

closed 2021-06-18 by tomeichlersmith