LDMX-Software/ldmx-sw #888: EventDisplay re-design

tomeichlersmith, 2020-08-18:
CMS uses a specific event display workflow that I think is really nice. Basically, their [actual event display](https://github.com/cms-outreach/ispy-webgl) is a JavaScript-based web application that takes json files as inputs. Then they have [analyzers that can translate the event bus objects](https://github.com/cms-outreach/ispy-analyzers) into the json files in a specific way that is readable by their web app.

I like this process for three reasons:
1. With the event display being a web-sever, it could be hosted within the container and visited by the user at a local IP address.
2. _All_ of the event object to display object translation is done separately from the drawing of the display. This is helpful because then the event display will be faster and it will be easier to debug (just open up the json to make sure the translation was done correctly).
3. If we want to change how an object is displayed, we don't touch the event display itself. Only the analyzer that does the translation.

The main downside of this process is that the display of the detector geometry is hard coded into the event display (like it currently is for us), but this issue may be rectified if I can find a method of translating gdml into the json files that the display reads in for the geometry.

tomeichlersmith, 2020-08-19:
I think we can use a more general, open-source project [Phoenix](https://github.com/HSF/phoenix) to base our event display on. It looks like it is [pretty active](https://github.com/HSF/phoenix/pulse) and takes `.obj` files as input for the geometry and `.json` files as input for the event objects.

tomeichlersmith, 2020-08-20:
It looks like we could convert our gdml into `.root` or `.json.gz` files that are importable by Phoenix:
- https://github.com/HSF/phoenix/issues/122
- https://github.com/HSF/phoenix/pull/117

tomeichlersmith, 2021-02-10:
Translation to json files from our objects in ROOT is necessary. I think an easy way to do this is with a python module.
https://chadrick-kwag.net/json-dumpingserializing-custom-python-classes/

omar-moreno, 2021-02-10:
Why not just build on what DD4hep has already done? Their geo display is
pretty nice.

tomeichlersmith, 2021-02-10:
I haven't looked into it...good idea

tomeichlersmith, 2021-02-10:
[The "documentation"](https://dd4hep.web.cern.ch/dd4hep/usermanuals/DDEveManual/DDEveManualse1.html#x2-10001) makes me worried, but I will play around with it when I get the chance.

omar-moreno, 2021-02-10:
Documentation is pretty bad ... but still might be worth over building from
scratch.  They seem to answer questions readily.

tomeichlersmith, 2021-02-10:
I guess, I'm happy to go either way and we need to just pick a path. Both paths have their pros and cons, so I want to lay them out here. I like number 2 better because the GUI doesn't use ROOT and actually looks cool, but the translation layer is a big con.

## 1. ROOT-based GUI

**Pros**
- Similar to our current approach
- Can be integrated with our geometry description and (potentially) use DD4hep directly
- Reads data files directly

**Cons**
- Slow
- Struggle to container-ize the application
  - This is because ROOT::Eve libraries don't compile without access to a display _during compilation_.

## 2. NPM-based GUI with Translation Layer

**Pros**
- GUI is fast and looks nice (not cluttered)
- Translation of ROOT objects to displayed objects is output to a file so it can be easily debugged
- Easily containerizable, the actual application is just viewed in a browser and the GUI handles importing the event file.

**Cons**
- Both the detector description and the event objects have to be translated to objects readable by Phoenix.

tomeichlersmith, 2023-06-15:
copying in my notes from slack chatting with @cmantill and @mrsolt 

---

The biggest frustration with developing a event visualization is the detector. How can we access the detector information so that it can be drawn and the visualization knows which detector cell to highlight for a specific hit. The "easy" way to get a visualized detector is to use G4 and the full GDML description but as you point out, this is too detailed. For the event display, we don't want to see all the individual material layers in the ECal, for example, since that just clutters the view. Other experiments have switched to a two-stage model where the processing framework (ldmx-sw in our case) writes out a new file (e.g. JSON) that only contains the objects that we wish to draw and then have a different program based on a language better-suited for visualization (e.g. javascript) load this information. This issue links to a project we could use called Phoenix that is this second program. Then the work for us would be two stages (1) getting our detector geometry read-able by this program and (2) writing processors that can convert our hits/tracks/particles into the objects to be drawn and write them to a JSON file.

1. I think the best idea would be to have a "condenser" program which writes a simplified geometry that can be loaded by the event display using G4's GDML reader to parse the full geometry and then re-write only the parts we care about. This would be hopefully fast enough to run that geometry changes (like v13->v14) would only require a quick re-run of this program.
2. This is probably simpler in terms of understanding the API but there are a lot of bugs lurking in order to convert our objects into the objects to be drawn. For example, the ecal hits would best be understood as tiny little hexagons but the hcal hits would be long bars oriented properly. This is where the geometry conditions would come into play.

> how bad of an idea is to make [a summer student] develop a base for event visualization for ldmx-sw?

I think this project is only suited to someone already familiar with C++. The complexities here would require them to be not-confused by C++ syntactical issues so that they can focus on the geometry complexities. Understanding JS and NPM so they could more easily interface with Phoenix would be a plus but I don't think is necessary since Phoenix is already a finished product (more or less, we just need to plug in our geometry and produce the JSON event files).

> Do you know if there has been any event visualization done before?

Yes. There was a ROOT TEve event display which attempted to be able to run LDMX event ROOT files (you can still see the ancient ruins of this display on trunk under the EventDisplay directory). The reason I abandoned it is because there was no automatic loading of the detector geometry and so the geometry needed to be manually updated (and I hate ROOT). After I abandoned this (and even slightly before since it was so awful to use - crashing after too many events regularly), many folks started using other visualization techniques which display various options - the process I describe above is the one I think would be easiest and longest-lasting.

1. G4 - with some time spent with the visualization manual, we could coax G4 into reading our ROOT files and drawing our objects. I fear the amount of work this would take is high; however, it may be worthwhile since then we would get the automatic geometry conversion "for free".
3. Matplotlib - basically requires the user to manually draw the detector elements they wish to be viewed. Not horrendous in our case since our detector is relatively simple, but still bothersome. I believe this would be hard to maintain since there is still no automatic geometry conversion.
4. NPM/Phoenix - what I described above. Erik Walin used this for the 8GeV paper event vis, although he mentioned he just hard-coded the geometry (and so there is still no conversion/condenser tool) but he could be a potential resource to help get an understanding on how to use Phoenix (or a Phoenix-like program).

tomeichlersmith, 2023-06-15:
Potential program for converting geometry: https://bitbucket.org/jairhul/pyg4ometry/src/develop/

AnmolS1Z, 2023-06-15:
Plan to create Event Display:

1. Convert the gdml files to root - matrices and parameterizations aren't easily converted so we'll need to do some extra work with them
2. Convert those root files to gltf files
3. Insert the gltf files into Phoenix to create display
4. Once everything works, can start writing event data.

AnmolS1Z, 2023-06-27:
You can skip steps 1 and 2 if you already have a root file.
Converting from gdml to gltf:
1) Remove any loops, array brackets, or references to other gdml files in the file you want to convert. This means flattening the loops, replacing the array bracket with the exact element from the array; So for the array "[1,2,3,4]", replace array[1] with 1, since gdml uses 1 based indexing, and replacing entity calls with the contents of the file.

2) Convert gdml to root with gdml_to_root_export.cxx using the following command: "ldmx root -l gdml_to_root_export.cxx". Be sure the filename inside gdml_to_root_export.cxx is the correct gdml file

3) Use the following repo: https://github.com/HSF/root_cern-To_gltf-Exporter/tree/main, make an html file based off of export_LHCb.html, replacing filenames and arrays as needed. A working html file for the whole detector will be included as export.html.

4) The export.html file has 7 array for the parts it needs to generate. If you don't want to include a part, just comment out that line. This has been done for the magnet

5) Put the new gltf file into phoenix

AnmolS1Z, 2023-06-27:
Thank you. Could you provide the gdml to root code?

pbutti, 2023-06-27:
Sure!
If you don't need it urgently I'll just put it on github tomorrow with instructions.

tvami, 2025-01-24:
This tool https://github.com/vmc-project/vgm/tree/v5-2?tab=readme-ov-file
seems good too

tvami, 2025-01-24:
well I guess I should add that it seems the most annoying / less robust / not straightforward part is to go from the GDML to ROOT, since in Anmol's work the first step is to flatten loops and replace the array bracket, so we should find a way that can just get the GDML as input and have the ROOT / GLTF as output. And for that the VMS above could be useful, I think