LDMX-Software/ldmx-sw #899: Error in v2.2.0 when low number of hits in HCAL (?)

sophiemiddleton, 2020-09-02:
**Describe the bug**

v2.2.0 works fine when a large number of hits. But in a sample with fewer/no hits I see this error within the simulation:

```
Error in <TBranchElement::TBranch::Fill>: Failed to write out basket.

Error in <TBranchElement::Fill>: Failed filling branch:HcalSimHits_v12.timeContribs_, nbytes=-1
Error in <TBranchElement::Fill>: Failed filling branch:HcalSimHits_v12.HcalSimHits_v12.timeContribs_, nbytes=-1
Error in <TBranchElement::TBranch::WriteBasketImpl>: basket's WriteBuffer failed.

Error in <TBranchElement::TBranch::Fill>: Failed to write out basket.

Error in <TBranchElement::Fill>: Failed filling branch:HcalSimHits_v12.nContribs_, nbytes=-1
Error in <TBranchElement::Fill>: Failed filling branch:HcalSimHits_v12.HcalSimHits_v12.nContribs_, nbytes=-1
Error in <TTree::Fill>: Failed filling branch:LDMX_Events.HcalSimHits_v12, nbytes=-1, entry=8244
 This error is symptomatic of a Tree created as a memory-resident Tree
 Instead of doing:
    TTree *T = new TTree(...)
    TFile *f = new TFile(...)
 you should do:
    TFile *f = new TFile(...)
    TTree *T = new TTree(...)

Error in <TBranchElement::TBranch::WriteBasketImpl>: basket's WriteBuffer failed.

Error in <TBranchElement::TBranch::Fill>: Failed to write out basket.

Error in <TBranchElement::Fill>: Failed filling branch:samples_, nbytes=-1
Error in <TBranchElement::Fill>: Failed filling branch:EcalDigis_v12.samples_, nbytes=-1
Error in <TTree::Fill>: Failed filling branch:LDMX_Events.EcalDigis_v12, nbytes=-1, entry=8251
 This error is symptomatic of a Tree created as a memory-resident Tree
 Instead of doing:
    TTree *T = new TTree(...)
    TFile *f = new TFile(...)
 you should do:
    TFile *f = new TFile(...)
    TTree *T = new TTree(...)

Error in <TBranchElement::TBranch::WriteBasketImpl>: basket's WriteBuffer failed.

Error in <TBranchElement::TBranch::Fill>: Failed to write out basket.

Error in <TBranchElement::Fill>: Failed filling branch:HcalRecHits_v12.id_, nbytes=-1
Error in <TBranchElement::Fill>: Failed filling branch:HcalRecHits_v12.HcalRecHits_v12.id_, nbytes=-1
Error in <TBranchElement::TBranch::WriteBasketImpl>: basket's WriteBuffer failed.

Error in <TBranchElement::TBranch::Fill>: Failed to write out basket.

Error in <TBranchElement::Fill>: Failed filling branch:HcalRecHits_v12.amplitude_, nbytes=-1
Error in <TBranchElement::Fill>: Failed filling branch:HcalRecHits_v12.HcalRecHits_v12.amplitude_, nbytes=-1
Error in <TBranchElement::TBranch::WriteBasketImpl>: basket's WriteBuffer failed.

Error in <TBranchElement::TBranch::Fill>: Failed to write out basket.

Error in <TBranchElement::Fill>: Failed filling branch:HcalRecHits_v12.energy_, nbytes=-1
Error in <TBranchElement::Fill>: Failed filling branch:HcalRecHits_v12.HcalRecHits_v12.energy_, nbytes=-1
Error in <TBranchElement::TBranch::WriteBasketImpl>: basket's WriteBuffer failed.

Error in <TBranchElement::TBranch::Fill>: Failed to write out basket.

Error in <TBranchElement::Fill>: Failed filling branch:HcalRecHits_v12.time_, nbytes=-1
Error in <TBranchElement::Fill>: Failed filling branch:HcalRecHits_v12.HcalRecHits_v12.time_, nbytes=-1
Error in <TBranchElement::TBranch::WriteBasketImpl>: basket's WriteBuffer failed.

Error in <TBranchElement::TBranch::Fill>: Failed to write out basket.

Error in <TBranchElement::Fill>: Failed filling branch:HcalRecHits_v12.xpos_, nbytes=-1
Error in <TBranchElement::Fill>: Failed filling branch:HcalRecHits_v12.HcalRecHits_v12.xpos_, nbytes=-1
Error in <TBranchElement::TBranch::WriteBasketImpl>: basket's WriteBuffer failed.

Error in <TBranchElement::TBranch::Fill>: Failed to write out basket.

Error in <TBranchElement::Fill>: Failed filling branch:HcalRecHits_v12.ypos_, nbytes=-1
Error in <TBranchElement::Fill>: Failed filling branch:HcalRecHits_v12.HcalRecHits_v12.ypos_, nbytes=-1
Error in <TBranchElement::TBranch::WriteBasketImpl>: basket's WriteBuffer failed.

Error in <TBranchElement::TBranch::Fill>: Failed to write out basket.

Error in <TBranchElement::Fill>: Failed filling branch:HcalRecHits_v12.zpos_, nbytes=-1
Error in <TBranchElement::Fill>: Failed filling branch:HcalRecHits_v12.HcalRecHits_v12.zpos_, nbytes=-1
Error in <TBranchElement::TBranch::WriteBasketImpl>: basket's WriteBuffer failed.

Error in <TBranchElement::TBranch::Fill>: Failed to write out basket.

Error in <TBranchElement::Fill>: Failed filling branch:HcalRecHits_v12.pe_, nbytes=-1
Error in <TBranchElement::Fill>: Failed filling branch:HcalRecHits_v12.HcalRecHits_v12.pe_, nbytes=-1
Error in <TBranchElement::TBranch::WriteBasketImpl>: basket's WriteBuffer failed.

Error in <TBranchElement::TBranch::Fill>: Failed to write out basket.

Error in <TBranchElement::Fill>: Failed filling branch:HcalRecHits_v12.minpe_, nbytes=-1
Error in <TBranchElement::Fill>: Failed filling branch:HcalRecHits_v12.HcalRecHits_v12.minpe_, nbytes=-1
Error in <TTree::Fill>: Failed filling branch:LDMX_Events.HcalRecHits_v12, nbytes=-1, entry=8339
 This error is symptomatic of a Tree created as a memory-resident Tree
 Instead of doing:
    TTree *T = new TTree(...)
    TFile *f = new TFile(...)
 you should do:
    TFile *f = new TFile(...)
    TTree *T = new TTree(...)

........````

**To Reproduce**
Steps to reproduce the behavior:
1. Go to /nfs/g/slac/ldmx/users/smidd/WAS/*.lhe (use any)
2. Run a simulation and attempt to reconstruct
4. See error

**Desired behavior**
A clear and concise description of what you want to happen (i.e. instead of the bug).

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment:**
 - Inside the container:
    - Output of `ldmx-container-config`
 - Outside the container:
    - OS of computer you are running on
    - Version of Geant4, ROOT, ONNXRuntime, Xerces, and gcc

**Additional context**
Add any other context about the problem here.

closed 2020-09-02 by sophiemiddleton

tomeichlersmith, 2020-09-02:
What ended up being the fix for this?

bechenard, 2020-09-02:
And what was the problem? Was planning to look at it tonight but if it's 
fixed...

Bertrand