LDMX-Software/ldmx-sw #904: ECal Layer Shifting and incorporate layerID in ECalHexReadout

Hongyin-debug, 2020-09-07:
**Is your feature request related to a problem? Please describe.**
Layer-shift the second layer within each bi-layer of Ecal layers--- to cover cracks.
ECalHexReadout is a x-y map of cells, but does not describe how x and y changes with z.
ECalID.h has layers defined in terms of double layers (i.e. 17 layers with 12 ring modules in a layer)--- not convenient for layer shifting (need to separately define 34 layers)

**Describe the solution you'd like**
Need to add layerID and incorporate z as a function of x-y in ECalHexReadout.h and .cxx.
Add Layer shifting in ECalHexReadout.
Change ECalID so that there are 34 layers and 6 ring modules in each.

**Describe alternatives you've considered**
Link the ecal.gdml file to ECalHexReadout so that it's not necessary to change ECalHexReadout every time a new layer-configuration is tried out?

**Additional context**
[image]

jmmans, 2020-09-07:
Dear Hongyin,

EcalID doesn't know how many layers there are, it is simply a container 
for values passed in to it.  The assignment of layer numbers is done in 
EcalHexReadout as used in the simulation code.

I agree with the idea that each actual layer should be separate in the 
code, so if it's true that we have double layers in there now, it would 
make sense to change it.  This will break a lot of things, however, so 
it cannot be done in the mainline until a future major release.  It 
would be reasonable to pursue as a branch development until then.

Regarding the GDML and EcalHexReadout issue -- this is a brittle 
connection because the GDML is only loaded when an actual simulation job 
is run.  If the digis are stored and then loaded back off disk, the GDML 
isn't loaded.  Instead, EcalHexReadout needs to know what to do 
_without_ the GDML available.  Therefore, the necessary information 
would need, for example, to be put into the RunHeader.  It could be 
sourced from the GDML.  Detailed proposals of what information is 
sufficient would be valuable.

J. Mans

Hongyin-debug, 2020-09-07:
Hi Jeremiah,
I see! Yes, the ECal ID has container values of up to 17 layers.
In the original ECalHexReadout, the position map of the modules are in terms of x-y, so every layer is a copy of the x-y map of the first layer (or zeroth layer).  So in the original ECalHexReadout, there isn't really an assignment of layers numbers in terms of double layers, it's just it doesn't have each layer (or double layer) uniquely defined at all. 
So originally, the key value for modulePositionMap_ only has moduleID, I'm thinking of changing it to a pair as key (<layer ID, moduleID>), and adding some z positions in ECalHexReadout.
The branch iss904 focuses on this issue, but it's still in progress. I'll send around a message when I think it's ready for double-checking.
The branch I'm currently working on has the layer shifting incorporated, do you think it's necessary to start a separate branch just for adding layerID into ECalHexReadout without the layer shifting?
-Hongyin

tomeichlersmith, 2020-09-08:
Hi Hongyin,
I'm not sure if your interpretation of the IDs is correct. How are you determining the maximum is 17? The layer and module IDs are decoded from the copy number in ecal.gdml as described [in this document](https://www.overleaf.com/read/hfhrgtyftykd).

Another suggestion I have is to avoid creating another data structure. you could use `EcalID(layerID,moduleID,0)` to be the key in your module position map instead of a new combination of of layer and module IDs. This is what we do for the "flattened" map where we don't include the layer ID, so you could think of this as another "flattening" in the "cell ID" axis.

Lastly, I don't think that it is necessary to have a separate branch for adding the layer ID into EcalHexReadout without the layer shifting. As long as you can set the layer shift to zero (and recover the original layout) at run time (using python configuration variables).
Tom

Hongyin-debug, 2020-09-09:
Hi Tom,
Yes, everything seems fine in ecal.gdml. For ECalID.h though, it shows:
	static const RawValue LAYER_MASK{0x3F}; // space for up to 64 layers
	static const RawValue LAYER_SHIFT{17};
	static const RawValue MODULE_MASK{0x1F}; // space for up to 32 modules/layer
	static const RawValue MODULE_SHIFT{12};
I'm thinking of changing it to:
	static const RawValue LAYER_MASK{0x3F}; // space for up to 64 layers
	static const RawValue LAYER_SHIFT{34};
	static const RawValue MODULE_MASK{0x1F}; // space for up to 32 modules/layer
	static const RawValue MODULE_SHIFT{7};
I'm not sure if this will affect the other files much, but I'm guessing not?

And yes, it is a good idea to use instead EcalID(layerID,moduleID,0) as key since it would already include both layerID and moduleID.

I see, yes, the layer shift value is easy to set :D 
Hongyin

jmmans, 2020-09-09:
Dear Hongyin,

This MUST NOT BE CHANGED.  It will horribly break data on disk.

The "LAYER_SHIFT" has nothing to do with the number of layers allowed. 
It has to do with how the 32-bit unsigned integer is decoded.

To get the layer out of the id, you take the 32-bit integer, shift it 
right by 17 bits and then apply an AND mask to it to select the bottom 
six bits.  This is all done internally.

Notice the comment -- we can store up to 64 layers, 32  modules/layer, 
etc.  This is to allow for possible future studies.  In _reality_, the 
id current is used up to 34 layers, 7 modules per layer.

Jeremy

Hongyin-debug, 2020-09-09:
Hi Jeremiah,
That makes sense!
-Hongyin

Hongyin-debug, 2020-09-17:
The submodules for this issue can be found here:
Ecal: https://github.com/LDMX-Software/Ecal/tree/iss904
SimCore: https://github.com/LDMX-Software/SimCore/tree/iss904

Hongyin-debug, 2021-01-27:
A new branch has been set up for the rotated Ecal geometry.
Changes are in EcalHexReadout.cxx and Ecal.gdml.
[Ecal Rotation.pdf](https://github.com/LDMX-Software/ldmx-sw/files/5880463/Ecal.Rotation.pdf)

PR #1079 (2022-09-09): ECal Portion of DR Geometry Update
I am updating _ldmx-sw_, here are the details.

**This is a draft PR because I have not tested the GDML updates in simulation; however, I think I've implemented the complicated C++ side of things inside of `EcalGeometry` (formerly `EcalHexReadout`) and the visualization of the GDML detector looks correct.**

### What are the issues that this addresses?
- resolves #904
- resolves #923 
- resolves #890 

## Check List
- [x] I successfully compiled _ldmx-sw_ with my developments
- [x] I ran my developments and the following shows that they are successful.
- [x]  [... cut]
closed 2022-09-09 by tomeichlersmith