LDMX-Software/ldmx-sw #906: v2.2.1 validation

vdutta, 2020-09-14:
Collecting here plots related to validating v2.2.1 samples.

Attached below are some plots related to the ECAL validation, comparing v2.2.1 ECAL sim hits to a v2.1.1 sample. All plots are normalized to 1. In general, the comparisons seem to agree with the changes I'm aware of, i.e. the hexagonal modules should now have 432 cells rather than 397, and the front z position is shifted back from 220 mm to 240.5 mm to agree with what's currently in the gdml. @tomeichlersmith I had a question about the 397 -> 432 cell change: on slide 2 it seems the hexagons in the readout are no longer quite regular, is that the case, or is it just an artefact of how I'm drawing the boundaries (by hand :)), and perhaps the cell-center - to - edge space is just larger on some sides than others?
[v2.2.1_validation_ecal_091220.pdf](https://github.com/LDMX-Software/ldmx-sw/files/5219386/v2.2.1_validation_ecal_091220.pdf)

jmmans, 2020-09-14:
Dear Valentina,

Indeed, the actual geometry of the detector is such that three sides of 
the sensor have "short cells" and the other three sides have "long cells".

For final reconstruction purposes, we may want to consider algorithms to 
combine the edge/boundary cells between modules into "effective cells" 
with corrections for dead meaterial.

Jeremy

vdutta, 2020-09-14:
@jmmans thanks for the clarification! Indeed, a had a look again at the sensor diagram and it makes sense now :)

bryngemark, 2020-09-17:
I ran over runs 0-10 in 
`/nfs/slac/g/ldmx/data/mc/v12/4gev_1e_ecal_pn_val/4gev_1e_ecal_pn_v12_ldmx-det-v12_run*.root`
with the singularity image 
`/nfs/slac/g/ldmx/production/singularityImages/ldmx-v2.2.1-gLDMX.10.2.3_v0.4-r6.20.00-onnx1.3.0-xerces3.2.3-ubuntu18.04.sif`

with a run script that (apart from input and utput file arguments etc) had these relevant lines:
```from LDMX.DQM import dqm
trigScintTag= dqm.TrigScintSimDQM("TrigScintDQMTag")
trigScintTag.hit_collection = "TriggerPadTaggerSimHits"
trigScintTag.pad = "tag"
```

and similar for `Up` and `Down`.

The hit ID distribution looks like it undulates more than before?

[image]

and I'm not sure it's reflected in the y distribution (which is the direction the bar segmentation is):
[image]

x distribution looks ok
[image]

**But** it looks like something has happened to hit->getEdep() because those histograms are empty.

Rest looks trivially ok.

closed 2020-10-15 by omar-moreno