LDMX-Software/ldmx-sw #921: Add HCal as submodule

omar-moreno, 2020-10-05:
All existing HCal code should be moved to the submodule and then pulled into ldmx-sw as a submodule.

tomeichlersmith, 2020-10-09:
I've been working with @cmantill and @nhanvtran to update the digis and I just want to say that they are working on a lot of updates to make it *a lot* more realistic. With this in mind, @cmantill I think it would be helpful for you to make the `main` branch on your Hcal repo the old digi producer and old veto and then after a separate branch for your updates. That way you don't have to rush on your updates and we can still move Hcal to its own repo.

Also, there will be a lot of other structural changes (i.e. files going in different places) that we will need to transfer to the Hcal repo, and it will be nice to have the digi stuff on a separate branch until we get comfortable with the new structure.

cmantill, 2020-10-09:
hi @tomeichlersmith, I think that suggestion makes sense. I made the `main` branch the one with the old producers and will continue working on `cmantill/hcaladc` on the updates on the digi. 
Should I make a PR for https://github.com/LDMX-Software/ldmx-sw/tree/iss921?
Is there any other Hcal code that needs to be moved here?

tomeichlersmith, 2020-10-12:
Yes, please make a PR and No, I can't think of any other processors that are hcal-related.

closed 2020-10-12 by omar-moreno