LDMX-Software/ldmx-sw #922: Allow setting of log level for a processor

omar-moreno, 2020-10-07:
**Is your feature request related to a problem? Please describe.**
Currently, the log level can only be set globally.  This is an issue when trying to debug a processor because messages from all other processors are also printed.  It would be useful to set the log level such that it only applies to a given processor.

tomeichlersmith, 2020-10-07:
@bryngemark I can handle the core framework stuff and loop you in after that is completed. I won't start on this until after the structural changes for v3.0.0 go in.

bryngemark, 2020-10-07:
@tomeichlersmith Sounds good!

tomeichlersmith, 2024-08-29:
https://github.com/LDMX-Software/ldmx-sw/issues/1409#issuecomment-2311040238

tomeichlersmith, 2024-08-29:
This is related to #1301 - in the sense that both would touch the same code files.

I would also be interested in moving `logging::open` as early as possible, specifically before any of the processors are created so that the `ldmx_log` messages are directed to the file/terminal as configured. Right now, `logging::open` doesn't happen until after the processors are `configure`d which means the logging messages in `configure` overrides are not formatted the same way and are not directed to a log file if one is specified.

PR #1458 (2024-09-17): logging fix and the long-awaited setting log-level by processor
These developments do the following
- interpret the message's logging level into a human readable string for easier parsing
- add event number to logging messages via a custom Formatter which resolves #1301 
- allow logging levels to differ between different channels by implementing a custom Filter which resolves #922 
- update the configuration so that the logging has its own configuration class, keeping its parameters organized and separate from the Process (the current syntax is supported by overriding `__getattr__` so this doe [... cut]
closed 2024-09-17 by tomeichlersmith