LDMX-Software/ldmx-sw #923: Rename and Clean-Up EcalHexReadout

tomeichlersmith, 2020-10-08:
EcalHexReadout is a class that we are now using for almost all translation between EcalID and real-space position, not just cutting up the module hexagons into cellular hexagons. With this in mind, it makes a lot more sense to call it something more general than `EcalHexReadout`, probably `EcalGeometry` would work best, and then it would line up with the provider that provides it `EcalGeometryProvider`.

I'm also going to take this opportunity to clean-up the Ecal precision digis and the HgcrocEmulator.

@omar-moreno I would love to put all of this Ecal-related stuff in the Ecal module. After the cmake changes you are working on, how difficult would it be to make a "Geometry" sub-module of Ecal (similar to how we are making "Event" submodules)? This way, `SimCore` and `Ecal::Ecal` could link to `Ecal::Geometry` and the source for `Ecal::Geometry` would be in the `Ecal` module.

omar-moreno, 2020-10-08:
> EcalHexReadout is a class that we are now using for almost all translation
> between EcalID and real-space position, not just cutting up the module
> hexagons into cellular hexagons. With this in mind, [...]
This sounds like a good idea. I have already started doing something
similar in SimCore (added SimCore/Geo) and plan to do the same in the
Tracking module.

tomeichlersmith, 2021-06-18:
> Moving future updates from #873 here.

### Known un-realistic issues to rectify:

- [ ] The cell ID with respect to the x-coordinate will flip between sensitive layers because the modules will be flipped when mounted to make the full layer "sandwich".
- [ ]  Mousebites: The cells do not penetrate into the corners of the module hexagon to leave room for mounting

### Current known layout of module:

[image]

PR #1079 (2022-09-09): ECal Portion of DR Geometry Update
I am updating _ldmx-sw_, here are the details.

**This is a draft PR because I have not tested the GDML updates in simulation; however, I think I've implemented the complicated C++ side of things inside of `EcalGeometry` (formerly `EcalHexReadout`) and the visualization of the GDML detector looks correct.**

### What are the issues that this addresses?
- resolves #904
- resolves #923 
- resolves #890 

## Check List
- [x] I successfully compiled _ldmx-sw_ with my developments
- [x] I ran my developments and the following shows that they are successful.
- [x]  [... cut]
closed 2022-09-09 by tomeichlersmith