LDMX-Software/ldmx-sw #932: v2.3.0 Validation

omar-moreno, 2020-10-22:


omar-moreno, 2020-10-22:
Validation samples: /nfs/slac/g/ldmx/data/mc/v12/4gev_1e_pn_recon/batch1

vdutta, 2020-10-22:
From a first look at a handful of files, it looks like the total reconstructed energy in the ECAL is now considerably lower compared to v2.1. @tomeichlersmith is this expected? I thought the mean should be roughly the same?

[summedDet_base.pdf](https://github.com/LDMX-Software/ldmx-sw/files/5426417/summedDet_base.pdf)

jmmans, 2020-10-22:
Did the sensor thickness change in the simulation?

vdutta, 2020-10-23:
Hmm, no ... I don't believe anything's changed in the gdml.

tomeichlersmith, 2020-10-23:
*Software*: v2.1.1 had set `isNoise` to true to _all_ rec hits because of my own forgetfulness, but v2.3 has re-implemented it, so this shift _could_ be from including/not-including noise hits. This would account for the down shift _and_ the overall tightening of the distribution. You could look at sim-level plots to make sure that this is true. You could also check the v2.1.1 rec hits for pure noise hits by comparing det IDs to the sim hits (like I do [here](https://github.com/LDMX-Software/Ecal/blob/63703d75d62fd57562c33a9ba38cc8ea65d527fd/src/EcalRecProducer.cxx#L154))

*Physics*: I'm thinking this _could_ be from the detector shifting downstream (and therefore missing some wide particles).
A few plots that could verify this hypothesis:
- number of readout hits: if the downstream shift is losing wide particles we would also see a lower number of hits
- total sim eDep and total sim number hits: just as a sanity check to make sure we aren't doing anything funky in the rec pipeline. The shifts here should match the shifts in the rec hits if this hypothesis is correct.

jmmans, 2020-10-23:
Dear Tom,

I doubt both.  The narrowing is likely just a propotional effect from 
the reduced scale -- everything scales down together.  Also, the shift 
of the detector would affect tails, not the bulk of the distribution.

I strongly suggest making a plot of the deposited simulation energy. 
That will separate effects due to digi/reco from geometry effects.

Jeremy

vdutta, 2020-10-23:
@tomeichlersmith @jmmans I hope the attached plots help to clarify things. The sim hits are basically unchanged, it is the rec hit energies that seem to have shifted downwards. The effect of the downstream shift can be seen in the second set of plots, which shows a comparison between v2.2.1 and v2.1.1 for sim hits: other than the shift in z positions, it doesn't really affect the energy distributions and only has a slight effect on the total number of hits.
[v2.3.0_validation_102320.pdf](https://github.com/LDMX-Software/ldmx-sw/files/5430680/v2.3.0_validation_102320.pdf)

tomeichlersmith, 2020-10-23:
More thoughts:
- I have actually implemented a "readout threshold" in the digitization pipeline, so some very low energy sim hits will be dropped. This could account for the small down-shift in number of readout hits.
- For the ADC readout mode (E < ~10MeV), I don't get the actual voltage amplitude, I just use the maximum voltage measurement. Since this maximum measurement is always <= the actual voltage amplitude, this would account for the down-shift in energy deposited in a cell for these hits. I'm not really sure how to fix this? Maybe just need to update the calibration constants to account for this down-shift when reconstructing?

jmmans, 2020-10-23:
Hi Tom,

(1) I assume this is based on the sum of the simhits in a given chanenl 
(the fully-merged hit).

(2) Is this a reco or digi step?

Jeremy

tomeichlersmith, 2020-10-23:
(1) Yes, so I've been assuming that it has a negligible effect.
(2) Reco

bryngemark, 2020-10-26:
I’ve checked the trigger scintillator reco hits using the DQM histograms and they all look alright. I haven’t gotten to checking clusters and tracks; I’m on vacation this week. Anyway it looks like there is still a lot to understand for ECal but if @awhitbeck happens to have time to validate clusters and tracks this week then please do.

omar-moreno, 2020-10-27:
First round of Photo-nuclear validation plots: https://github.com/omar-moreno/ldmx-notebooks/blob/master/pn_validation/ecal_photonuclear_validation_v12_v2.3.0.ipynb

Things look as expected.

bechenard, 2020-10-30:
Hi guys,
Here are the validation plots for the HCal, and the comparison with a previous version of the code (don't remember which one exactly....). The plots make sense, but I do see more activity in the back HCal and less in the side HCal w.r.t the previous code. Not sure why this is the case, I need to investigate.

[val.pdf](https://github.com/LDMX-Software/ldmx-sw/files/5462669/val.pdf)
[valcomp.pdf](https://github.com/LDMX-Software/ldmx-sw/files/5462670/valcomp.pdf)

tomeichlersmith, 2020-11-02:
Investigating ECal issues revealed in validation at https://github.com/LDMX-Software/Ecal/issues/21
Branch is `iss932` to reference this validation issue.

awhitbeck, 2020-11-04:
I looked at a small fraction of the data.  Reconstructed clusters and tracks look reasonable.  Some distributions of clusters and tracks are shown below.  Things generally look fine.  @bryngemark I am not sure if there are other things you would like to check, though?

cluster locations (red: tagger, blue: upstream, black: downstream array)
[image]

clust multiplicities (same legend as above)
[image]

track location
[image]

track multiplicity: 
[image]

bryngemark, 2020-11-04:
@awhitbeck thanks, I agree this looks reasonable. To me this is enough; we should probably come up with a standard set of validation plots though and add them to the DQM module.

bryngemark, 2020-11-04:
For completeness, these are the reco TS hit DQM plots, on two files (run numbers 504 and 531). 

[image]
[image]
[image]
[image]
[image]
[image]

bechenard, 2020-11-13:
Hi all,

Apparently, my concerns were unfounded. It turns out I was comparing the latest recon with much older data than the I should have used - naming files 'old' and 'old2' seems to be a bad idea after all. I re-ran the validation script on the previous reco files, the results are available in the attached pdf. Basically, everything is fine. The bright side is that I expanded my validation script to look at more variables, this will hopefully prove useful in the future. 

In summary, the new reco files are fine for the HCal and I sign off

[valcomp.pdf](https://github.com/LDMX-Software/ldmx-sw/files/5534359/valcomp.pdf)

vdutta, 2020-11-13:
Attached are ECal validation plots created from the files in `/nfs/slac/g/ldmx/data/mc/v12/4gev_1e_pn_recon_iss932_v2`. While I see some small changes for very low energy hits compared to the previous version (v2.3.0-alpha), the overall downward shifts in the total reconstructed energy and number of hits are still there. (Side note: the v2.3.0-alpha comparisons are with a much smaller sample than the latest one). The shift between recon and sim hit energies still appears to be present for energy depositions of ~O(1 MeV) or higher. I'm not sure how to reconcile this with the plots @tomeichlersmith [posted](https://github.com/LDMX-Software/Ecal/pull/23). @omar-moreno is there any chance the changes were not propagated to the recon (long shot, I know)?
[v2.3.0_validation_111320.pdf](https://github.com/LDMX-Software/ldmx-sw/files/5539537/v2.3.0_validation_111320.pdf)

omar-moreno, 2020-11-13:
I just noticed that changes to Ecal/python/digi.py made a couple of days weren't picked up.  @tomeichlersmith, the change I'm referring to is `self.secondOrderEnergyCorrection = 4000. / 3983`. Do you expect this to significantly impact the recon?

tomeichlersmith, 2020-11-16:
That update would only shift the total rec energy towards 4GeV (up slightly) and would not change other aspects of the plots.

Here's the deal. v2.1.1 was _way_ too good because of how I unrealistically implemented it. The slight down offset in Sim Hit v Rec Hit distribution is expected from my non noise plots that @vdutta referenced. This effect is mainly originating from the fact that we are packing the DIGIs into 10-bit measurements and so (in effect) we are _always_ rounding down. Since all individual hits are slightly rounded down in this way, their sum is also below what it used to be. We can (1) try to account for this by increasing the reconstruction gain and/or (2) move forward with this set knowing that we made it more realistic.

tomeichlersmith, 2020-11-16:
Ok, I've updated some recon parameters in order to account for the DIGI truncation. In the ADC range, this increases the recon gain by 1% and in the TOT range, this increases the recon gain by 0.5%. This now means that (without the 2nd order energy correction), the mean of 100k 4GeV showers is now 4.023GeV (where before it was 4.010GeV).

I still want to advocate for the 2nd point I made above. These changes are drastic and comparing them to old versions of the ecal digi chain is complicated. I don't expect us to get the current DIGI chain to match the v2.1.1 DIGI chain.

omar-moreno, 2020-11-16:
Given that it's not an apples to apples comparison, point 2 is reasonable.
If there aren't any objections, I'll make a release today!

vdutta, 2020-11-16:
@tomeichlersmith understood and agreed that the comparison with previous (unrealistic) versions of recon is not fair. However, I would just like to make sure we understand the changes and are satisfied that we have a reasonable representation of the physics moving forward. There were a couple of questions which I hope you can help to me to understand:
1) The "EcalDigiVerify" plots that you posted [here](https://github.com/LDMX-Software/ldmx-sw/pull/934) show very good agreement between sim and recon hit amplitudes up to 10s of MeV. This doesn't really agree with the plot in my slide. Also, the total reconstructed energy that you show in inclusive events peaks still at ~4 GeV, which doesn't agree with an overall shift in the energy. Are the differences coming from differences between PN and inclusive events?
2) Do you understand the discontinuity in the "single cell energy" plot at ~500 MeV? Is there something specific in how the digi chain is implemented that causes this feature?

jmmans, 2020-11-16:
@vdutta , can you run inclusive events through your code setup?  It's hard to disentangle effects when we are looking at different classes of samples.  The "highest single cell energy" could be an effect from the TOT/ADC boundary, but we don't see this in the inclusive sample as far as I know (Can you confirm, @tomeichlersmith ?)

tomeichlersmith, 2020-11-16:
1. Yes, the PN events are expected to _not_ peak at 4GeV because a lot of the energy is "lost" to PN interactions. This is why I've been using the inclusive events to test the digi chain.

2. Does this use `getAmplitude` or `getEnergy`? (i.e. does this single cell energy include the layer weights?) There are a few places where discontinuities could pop up, and the layer weights mess with that interpretation.

Without the layer weights, there are two places that come to mind:
 - The ADC-TOT boundary: ~6.5MeV, this is where the chip starts reading out in TOT mode instead of ADC mode
 - The Low-High TOT boundary: ~27.8MeV, this is where the chip discards the lowest 3 bits of the 12-bit TOT measurement so that it can be packed into the 10-bit space in the DIGI sample

Using the layer weight in the middle of the ECal (9.630 MeV/MIP), these boundaries would shift up.
 - ADC-TOT: ~490MeV
 - Low-High TOT: >2000MeV

So including layer weights, I would guess that discontinuity is from the ADC-TOT boundary. This discontinuity _is expected_ because the TOT and ADC measurements are so different.

jmmans, 2020-11-16:
@tomeichlersmith -- Your argument about the discontinity doesn't hold water, I think.  The _physical_ distribution is certainly continuous.  Your digi plot doesn't show any discontinuity, therefore TOT/ADC doesn't explain the observation.  Can you reproduce this plot for our samples?

vdutta, 2020-11-16:
@jmmans ... yes, I can definitely run over some inclusive events with the same setup. I'm taking a look at the file here: `/nfs/slac/g/ldmx/users/eichl008/ecal-digi/ecal_sim_recon_dqm_100k.root` now unless there is a later sample to look at.

The "max single cell energy" is filled in the EcalVeto class and uses `getEnergy`, so including the layer weights.

vdutta, 2020-12-04:
Latest ECal validation plots made from the PN files @omar-moreno produced under `/nfs/slac/g/ldmx/data/mc/v12/4gev_1e_pn_recon_iss932_v4`. Everything now looks reasonable to me ... thanks @tomeichlersmith and @jmmans for all the investigation!
[v2.3.0_validation_120420.pdf](https://github.com/LDMX-Software/ldmx-sw/files/5646106/v2.3.0_validation_120420.pdf)

omar-moreno, 2020-12-05:
Alright, I think we are ready for a release.

closed 2020-12-07 by omar-moreno