LDMX-Software/ldmx-sw #937: Out Of Time (OOT) Pileup Simulation

tomeichlersmith, 2020-11-09:
**Is your feature request related to a problem? Please describe.**
OOT Pileup is expected to be a difficulty LDMX experiences relatively rarely in its early stages, but may experience more of it in later phases of LDMX with higher luminosity or a higher rate.

**Describe the solution you'd like**
OOT Pileup can be simulated in the most robust way by enhancing the OverlayProducer that has been implemented primarly to study In-Time Pileup (multielectrons). Besides modifying the OverlayProducer, we would also need to enhance the various subsystem digitization pipelines to correctly account for timing separation of the different simulated hits. Correctly incorporating timing information in the DIGI stages is expected to be the most difficult part of this development.

**Describe alternatives you've considered**
Other alternatives would be to look at defining OOT at the DIGI level. This would be lighter in terms of output file size, but would be a lot more complicated to emulate correctly, especially across the wide variety of our subsystems.

bryngemark, 2020-11-09:
Just a comment that the OverlayProducer already deals with OOT pileup to first order, in the sense that you can put the overlay event in bunches other than the sim event (sampled from a uniform distribution of +/- _m_ bunches). What it so far doesn't deal with is having no sim event *at all*. But it is already possible to look at a bunch other than the sim event bunch, or even artificially cut that bunch out based on time stamps if you would like to. 

So what might be cool to be able to do here, is to just have an inclusive multielectron sample spread out in time, with no interesting simevent in it. If overlaying a 1e inclusive electron sample with some small mu would work, then that's already possible with the current OverlayProducer.

For the digi work to correctly pick up on the timing information though, I agree completely.

tomeichlersmith, 2021-04-19:
At the digi-level, the updates to the HGC ROC emulator theoretically handles OOT pileup correctly. This needs to be validated, but the emulator's use in the HCal (which has a much broader time spread for IT hits) is a first-order validation of this. In any case, closing this because we have the sim and digi tools necessary to study OOT pileup.

closed 2021-04-19 by tomeichlersmith