LDMX-Software/ldmx-sw #943: TS tracking doesn't fully remove duplicates

bryngemark, 2020-11-20:
**Describe the bug**
There are cases where clusters in the two downstream pads are assigned to different tagger seed clusters, creating a duplicate track. This happens because the overlap removal logic assumes that these overlapping tracks will be neighbors in the list of tracks, which is not always the case.

I have a patch for this.

tomeichlersmith, 2021-06-18:
@bryngemark Is this patch now on `trunk` when we updated the TrigScint module to `trunk` with #1000 ? (It looks like it, but I'm less confident than the other issues I've been cleaning up).

bryngemark, 2021-06-18:
@tomeichlersmith yes it is! I just checked both the PR and then actually the code in trunk 😄 . I'll close this now.

closed 2021-06-18 by bryngemark