LDMX-Software/ldmx-sw #946: HGC ROC Emulation Improvement

tomeichlersmith, 2020-12-02:
_Context_: While @jmmans and I were debugging the weird behavior of the ECal reconstruction in the v2.3 validation samples, we talked about a lot of long-term improvements that "should" be made to the chip emulator but I didn't want to slow down the already late release.

### Timing Shift
The chip emulator should "know" what its time phase is relative to the simulation global time where t=0ns occurs when the primaries cross z=0mm. In the far future, this would be a conditions table assigning a time phase to each electronic ID corresponding to each chip. Since we aren't really using electronic IDs yet, there needs to be some middleground.

*Potential Solution*: I don't know of a better solution than having the ECal/HCal digi producers do the time shift themselves. They already need to change units from MeV to mV depending on their detector conditions, so this doesn't seem like too big of an ask. In this case, these comments are just warnings for @cmantill .

### Conditions Reliance
The chip emulator currently _can_ take input from a conditions table, but it should _require_ input from a conditions table. This would make HgcrocEmulator more closely resemble HgcrocTriggerCalculations and be more forward thinking.

*Potential Solution* I just need to do this. Not much else to it...

### TOA Measurement
Currently, the simulated time (after accounting for the time phase) is used as the time of the peak of the pulse. This means that for a lot of pulses, the TOA measurement (which is calculated as the time when the pulse goes over the TOA threshold) is negative. This is not realistic (obviously).

*Potential Solution* Use the sim time to set the peak time, then calculate the TOA, then shift the peak so that the calculated TOA corresponds with the actual sim time, then inject noise and actual digitize. 

### Other Improvements
Other improvements that can be looked at in this round of developing the emulator are
- Allow for shifting the SOI so that pre-samples can be taken effectively.
- Thoughtfully digitize multiple OOT hits (IT hits are more complicated because of the pre-amp shaping of the pulse)
- What do the ADC_t measurements look like during the TOT measurement (i.e. during saturation)? Right now, I'm just measuring a pulse, but I'm fairly certain that is inaccurate. Should it just rail at 1023?

jmmans, 2020-12-02:
Dear Tom,

Regarding the TOA --> the pulse should be created based on the Sim time, 
but this shouldn't necessarily be set to zero.  Actually, the ideal is 
to _center_ the pulse in the bunch window rather than have it arrive at 
the front edge.  In that case, the TOA should always be in the same 
bunch as the ADC (in the same readout/not negative).

Then, at the /reco/ stage, one needs to subtract the nominal TOA for 
in-time particles.

Jeremy

tomeichlersmith, 2021-03-12:
TOA measurement should only go into the sample where the pulse _actually_ went over the TOA threshold. The TOA's in the other samples should be a "non-measurement" of 0. The actual TOA measurement should be shifted up by one before being saved so that the "non-measurement" of 0 only happens when the TOA wasn't measured.

tomeichlersmith, 2021-03-22:
Following lines should be improved. Basically forces TOA to be greater than or equal to zero, which is not realistic.

https://github.com/LDMX-Software/ldmx-sw/blob/5593a7fd9f9e1043bc487cceb54bffc561431bc8/Tools/src/Tools/HgcrocEmulator.cxx#L200-L203

tomeichlersmith, 2021-04-19:
I forgot to link this issue to the recent PR #980  that was merged. I don't think that PR perfectly addressed all of these comments; however, it did address them and I think we need to re-focus any further development on mimic-ing an _actual_ chip.

closed 2021-04-19 by tomeichlersmith