LDMX-Software/ldmx-sw #955: Overlay producer validation

bryngemark, 2021-01-16:
This issue is for keeping track of the validation of the overlay producer. Please post results/comments here! 

The goal is to see that 
1. you can run what you usually run one these files, and get the expected results 
2. you get the same results from overlay as from a multiparticle gun sample

**What** 
10k events each of an "old way" multiparticle gun 2-electron sample, and a "new way" 1+1 electron sample produced using the pileup overlay processor (with no time shift).

**Where**
`/nfs/slac/g/ldmx/data/validation/v12/4gev_2e_inclusive/v2.3.0/`

**Note**
- make sure to use the collections with "overlay" in the name (the others contain only the 1e "sim" process that gets combined with the pileup hits)
- I can already now warn Hcal people that there are some hardwired collection names in certain processors, which I think prevent me from running the full chain of Hcal processors on sim hit collections produced in overlay. 
- If you find that you're unable to run your processors, please open and link to a separate github issue (and/or push a fix! :slightly_smiling_face:) so we can get it sorted.

tomeichlersmith, 2021-01-18:
Looking first and the ["lower level" ecal variables](https://github.com/LDMX-Software/ldmx-sw/files/5831191/low_level_ecal.pdf)
 (Digi encoded words and individual rec hits), the distributions mimic each other beautifully! The only separation is in the regions where there are less hits, but that is expected since we can't have these two samples be generated with the same random numbers at all steps. I will look next at the "higher level" ecal variables that get put into the BDT.

tomeichlersmith, 2021-01-18:
Again, the ["higher level" ecal variables](https://github.com/LDMX-Software/ldmx-sw/files/5831483/high_level_ecal.pdf) (BDT input variables) mimic each other in shape well. To me, it looks like the only separation is due to randomness; however, I would like @vdutta to glance at these plots as well.

vdutta, 2021-02-04:
@tomeichlersmith, sorry I missed your post! I confirm that I also looked at the files and compared the higher level variables and BDT output ... to me also everything seems to agree well enough. 
[overlay_validation_ecal_020321.pdf](https://github.com/LDMX-Software/ldmx-sw/files/5927771/overlay_validation_ecal_020321.pdf)

bryngemark, 2021-02-10:
@bechenard this is what we need Hcal to look at

bechenard, 2021-02-19:
@bryngemark Sorry for the delay. Just compared a few variables for the HCal and everything seems to agree well enough. I also looked at other 2d distributions and they are fine too (I can provide plots if you want)
[val1eover_2e_comp.pdf](https://github.com/LDMX-Software/ldmx-sw/files/6012827/val1eover_2e_comp.pdf)

tomeichlersmith, 2021-06-18:
@bryngemark How do you feel about this? It seems like we have found a patched various bugs, does that mean the overlay producer has been validated? If not, what is there left to do?

bryngemark, 2021-06-18:
Hi @tomeichlersmith, I think we've found what we can with small samples. The overall variables agree well between multi-e particle gun and 1e+pileup, and the spatial overlap was probably only going to be found in tracker or TS -- and we did. @GNiramay has validated pileup extensively for TS. So I think this is as much as we'll learn for now.

tomeichlersmith, 2021-06-18:
Sounds good :) I am going to close this and we can make a new issue if (when) bugs are found.

closed 2021-06-18 by tomeichlersmith